Seeed-Studio/PN532 #20: No tag is detected on Arduino Ethernet with ethernet enabled on D10 and NFC on D9.

MeanEYE, 2015-01-30:
NFC reader is initialized properly, but it never detects tag near it. When debugging is turned on I get this output:

    write:  4A 1 0
    Time out when waiting for ACK

There's probably an issue with `setBitOrder` but I was unable to find and verify it. Any ideas?

MeanEYE, 2015-01-30:
Managed to narrow it down a little bit. When ethernet adapter is enabled, everything goes well but it never arrives at [line 90 in PN532_SPI.cpp](https://github.com/Seeed-Studio/PN532/blob/master/PN532_SPI/PN532_SPI.cpp#L90).

MeanEYE, 2015-01-30:
Solved the issue by removing all `clock`, `endianess` and `mode` from `PN532_SPI.cpp` and creating `SPISettings` like this:

SPISettings nfc_settings(SPI_CLOCK_DIV8, LSBFIRST, SPI_MODE0);

    SPI.beginTransaction(nfc_settings);
    nfc.begin();
    nfc.SAMConfig();
    SPI.endTransaction();

It's important to call `SPI.begin()` before any of this code. Then every time you want to use NFC, you need to `beginTransaction` and end it.

xiongyihui, 2015-01-31:
@MeanEYE
Cloud you help to send a pull request to fix the issue? It will be appreciated.

MeanEYE, 2015-01-31:
I can modify it and make a pull request, but I am not sure about few things you use there. For example `read` and `write` functions used in that file. Are they part of SPI or something defined somewhere else?

xiongyihui, 2015-02-02:
Thanks. I see. We solve the issue in application code using your code above.

closed 2015-02-02 by xiongyihui