Seeed-Studio/PN532 #23: Cannot get past firmware check SPI

jlkalberer, 2015-03-12:
Hi,

I have a weird issue with the hardware I am using where I have to use SPI with software mode instead of the default hardware mode in this library.  You can see my changes here - https://github.com/jlkalberer/Tappt.Spark/blob/master/Source/PN532_SPI/PN532_SPI.cpp
The main changes are in the read/write functions.  I am targeting a Spark Core instead of Arduino.

Anyways, I am receiving this:
0 FF 6 FA D5 3 32 1 6 7 E8 0

When I do a request for the firmware.  This is the same thing that comes over the wire when I use a modified Adafruit library so it is the correct response (at least for software mode).

Could there be a difference between the default hardware SPI mode and the software SPI mode I added?

Does anyone have SPI set up with this so you can give me the output for the firmware response?

xiongyihui, 2015-03-12:
There should be one more '0x00' at the front.

jlkalberer, 2015-03-12:
Ok, it looks like I am always missing that first '0x00'.  I think it just has to do with the hardware using software SPI.  I have older code using the old NFC library and it was expecting '0x00 0xFF'.

Anyways, the next thing I ran into was [this line](https://github.com/Seeed-Studio/PN532/blob/master/PN532_SPI/PN532_SPI.cpp#L96).

command will always be equal to zero.  Shouldn't the line read:
uint8_t cmd = buf[0] + 1;

xiongyihui, 2015-03-13:
`command` should not be zero, it is set by [writeCommand](https://github.com/Seeed-Studio/PN532/blob/master/PN532_SPI/PN532_SPI.cpp#L44).

closed 2015-03-13 by jlkalberer