Seeed-Studio/PN532 #24: Emulate Tag

Arcorme, 2015-03-27:
Hello,

I use this shield (https://www.adafruit.com/product/789) with an Arduino Uno.

As it embedded a PN532 i used this soft with I2C connection to read tags, it works.

Now I would like to emulate a tag but it do not work, even if no error shows.

On Adafruit FAQ it is said that to emulate a NFC tag with PN532 you need a 'secure element' (https://learn.adafruit.com/adafruit-pn532-rfid-nfc/faq)

Can someone tell me what is this element please ? Why is it mandatory ? Why can't we just emulate a URL opening with the PN532 tag, why does it need a secure element ?

My use case is to use the emulation mode to first emulate a NFC tag connecting Android phone on Wifi and then change this tag to connect to a URL.

Thank you.

xiongyihui, 2015-03-28:
When using Arduino + PN532 to emulate a tag, you need to use SPI interface. I2C is not supported (See #15).

A secure element is not required to emulate a tag.

This library supports to emulate NFC Forum Type 4 tag. Get a look at the example - https://github.com/Seeed-Studio/PN532/blob/master/PN532/examples/emulate_tag_ndef/emulate_tag_ndef.ino

closed 2015-03-28 by xiongyihui

jaspreet77, 2016-10-20:
@xiongyihui the tag emulator code you suggested didn't work(using  it in spi mode(shorted the S0 and S1 pin)))