Seeed-Studio/PN532 #28: Only 128 bytes to receive

bigzed, 2015-06-26:
Hello,

I'm using your library to communicate with an Android. I can send a 228 byte long message from the PN532 to the android.
But the maximum amount of data I can receive is 128 bytes. Does this maybe be related to 128 bytes being the default MIU for llcp? And how can I change it? Hope you can help me, I'm lacking the specifications for llcp.

best,
Steve

xiongyihui, 2015-06-30:
We can use MIUX parameter to change MIU. When without a MIUX parameter, it use the default value 128. The MIUX may be probably added at [PN532::tgInitAsTarget()](https://github.com/Seeed-Studio/PN532/blob/master/PN532/PN532.cpp#L801), as described at 5.2.3 section of LLCP manual. I'm not sure what the format it should be.

[NFCForum-TS-LLCP_1_1.pdf](https://github.com/Seeed-Studio/PN532/files/392/NFCForum-TS-LLCP_1_1.pdf)

bigzed, 2015-07-02:
Thanks for your help! I opened a pull request with my chagnes. For now the MIUX is hardcoded into the command.

closed 2016-01-19 by xiongyihui

wshhja, 2022-08-01:
@bigzed hi,i'm working with a PN532 module to communicate with my android via p2p channel, i want to know how to read the PN532's data send by InASR,InPSL and InDataChange which know from AN133910.pdf.I have succeed with two PN532.but it seems absolutly diffrent between android and pn532.