Seeed-Studio/PN532 #30: Emulation not working properly

SamDecrock, 2015-07-13:
Hi,

When holding my phone against the NFC reader of my Arduino, my phone detects a card of type "JIS 6319-4" but with a serial number of "00:00:00:00:00:00:00:00". That doesn't seem right.

I enabled debugging messages in PN532_debug.h and this is what appears on the terminal when trying to read "the emulated card":

```
read:   8D 8 E0 80
write:  86
read:   87 29
status is not ok
tgGetData failed!
write:  52 0
read:   53 0
write:  8C 5 4 0 12 34 56 20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```

I'm going to try with another phone and report back if that works.

xiongyihui, 2015-07-17:
Did you get it work?

SamDecrock, 2015-07-17:
Well,

On another phone, it opens up a uri. As this was added as an NDEF message.
But when viewing the card info with an app, The serial number is still "00:00:00:00:00:00:00:00"

```
------- Emulate Tag --------
Ndef encoded message size: 29
SAMConfig
write:  14 1 14 1
read:   15
write:  8C 5 4 0 12 34 56 20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
read:   8D 8 E0 80
write:  86
read:   87 29
status is not ok
tgGetData failed!
write:  52 0
read:   53 0
write:  8C 5 4 0 12 34 56 20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
read:   8D 8 E0 80
write:  86
read:   87 0 0 A4 4 0 7 D2 76 0 0 85 1 1 0
write:  8E 90 0
read:   8F 0
write:  86
read:   87 0 0 A4 0 C 2 E1 3
write:  8E 90 0
read:   8F 0
write:  86
read:   87 0 0 B0 0 0 F
write:  8E 0 F 20 0 54 0 FF 4 6 E1 4 0 80 0 0 90 0
read:   8F 0
write:  86
read:   87 0 0 A4 0 C 2 E1 4
write:  8E 90 0
read:   8F 0
write:  86
read:   87 0 0 B0 0 0 2
write:  8E 0 1D 90 0
read:   8F 0
write:  86
read:   87 0 0 B0 0 2 1D
write:  8E D1 1 19 55 0 68 74 74 70 3A 2F 2F 77 77 77 2E 65 6C 65 63 68 6F 75 73 65 2E 63 6F 6D 90 0
read:   8F 0
write:  86
tgGetData failed!
write:  52 0
read:   53 0
write:  8C 5 4 0 12 34 56 20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
read:   8D 8 E0 80
write:  86
tgGetData failed!
write:  52 0
read:   53 0
write:  8C 5 4 0 12 34 56 20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
```

jlkalberer, 2015-08-30:
Yeah, I can't get this to work either.  I'm using SPI and here is my output:

write:  14 1 14 13338
read:   153338
write:  8C 5 4 0 0 0 0 20 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 03338
invalid 3

jlkalberer, 2016-02-12:
Did you ever figure this out?  I haven't done a readout of the commands yet but I had a couple of phones that didn't work with the emulation mode.

jlkalberer, 2016-05-06:
I may have a fix in #46

closed 2019-08-24 by Pillar1989

jlkalberer, 2019-08-24:
@Pillar1989 - was there a fix for this?