Seeed-Studio/PN532 #38: Turning on and off continues reading of UID

Netoperz, 2016-01-16:
Hello, I'm building a reader using 532 shield and i have one problem.
When card is in field the UID (using the code from iso14443a_uid sample)  the UID is printed to serial continuously , so as long as card is in range the same number is sent over and over. That makes problems when you want to fill in some field and card is hold longer than timeout.
So if the card stays on reader it will populate field with many the same UID one after another.

So how can i change or can that be changed in library, to have a different type of action:
If the card is initiated and stays on reader it's UID will be printed to serial only ones. And if you want to reread the card you just need to take it away and put ones again.

Any help would be appreciated because my knowledge of C is poor :)

I know that miguels library for RC522 behaves the way i would like the PN532 library to behave.

xiongyihui, 2016-01-17:
Hi, we can print the UID once and wait until the card is taken away.

```
#include <SPI.h>
#include <PN532_SPI.h>
#include "PN532.h"

PN532_SPI pn532spi(SPI, 10);
PN532 nfc(pn532spi);

void setup(void) {
  Serial.begin(115200);
  Serial.println("Hello!");

  nfc.begin();

  // Set the max number of retry attempts to read from a card
  // This prevents us from waiting forever for a card, which is
  // the default behaviour of the PN532.
  nfc.setPassiveActivationRetries(0xFF);
  
  // configure board to read RFID tags
  nfc.SAMConfig();
    
  Serial.println("Waiting for an ISO14443A card");
}

void loop(void) {
  boolean success;
  uint8_t uid[] = { 0, 0, 0, 0, 0, 0, 0 };  // Buffer to store the returned UID
  uint8_t uidLength;                        // Length of the UID (4 or 7 bytes depending on ISO14443A card type)
  
  // Wait for an ISO14443A type cards (Mifare, etc.).  When one is found
  success = nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, &uid[0], &uidLength);
  
  if (success) {
    Serial.println("Found a card!");
    Serial.print("UID Length: ");Serial.print(uidLength, DEC);Serial.println(" bytes");
    Serial.print("UID Value: ");
    for (uint8_t i=0; i < uidLength; i++) 
    {
      Serial.print(" 0x");Serial.print(uid[i], HEX); 
    }
    Serial.println("");
    
    // wait until the card is taken away
    while (!nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, &uid[0], &uidLength)) {
    
    }
  }
  else
  {
    // PN532 probably timed out waiting for a card
    Serial.println("Timed out waiting for a card");
  }
}
```

Netoperz, 2016-01-17:
In my case that code is not working the way it should. still prints the UID one after another if the card stays on reader.

xiongyihui, 2016-01-18:
Sorry, it should be `while (nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, &uid[0], &uidLength)) {}`

`!` should be removed.

[The example](https://github.com/Seeed-Studio/PN532/blob/master/PN532/examples/iso14443a_uid/iso14443a_uid.pde) is updated. Thanks for your suggestion :)

Netoperz, 2016-01-18:
Thank you, i thought that "!" is wrong. 
There is a timeout for waiting for next card to come, because when you read (initiate) one card, you cannot reread (reinitiate) on a shorter period than about 1500 ms , can i change that time ? is there a place to change that ?

Thank you for your support :)

xiongyihui, 2016-01-19:
The `bool readPassiveTargetID(uint8_t cardbaudrate, uint8_t *uid, uint8_t *uidLength, uint16_t timeout = 1000, bool inlist = false);` has a default timeout.

We can change the timeout, for example

`nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, &uid[0], &uidLength, 3000)`

closed 2016-01-19 by xiongyihui