Seeed-Studio/PN532 #39: Low power consumption mode?

shayo, 2016-01-28:
Hi,

Here is a question for whoever managed to get a grip over the long data sheet (I'm a bit lost there).
What is the optimal way to use this library to run the PN532 in ultra low power consumption mode?
Is there a way to turn on the transmitter just before checking if there is a card, and then turning it off?
How about using the library as is, but shutting down using RSTPD_N? How long does it take for the chip to "wake up" ?

Thanks!

Shay

Netoperz, 2016-01-28:
I was using that feature some time ago, the "wake up" works without delay, there is a way to send a high, or low state for the one of the PN532 pins to enable or disable the IC.

Pillar1989, 2019-08-24:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-24 by Pillar1989