Seeed-Studio/PN532 #40: Library Error: Android_HCE

ConorOD, 2016-02-04:
This is my error

`C:\Program Files\Arduino\libraries\PN532\emulatetag.cpp:49:16: error: expected identifier before numeric constant

C:\Program Files\Arduino\libraries\PN532\emulatetag.cpp:49:16: error: expected '}' before numeric constant

C:\Program Files\Arduino\libraries\PN532\emulatetag.cpp:49:16: error: expected unqualified-id before numeric constant

C:\Program Files\Arduino\libraries\PN532\emulatetag.cpp:49:31: error: expected declaration before '}' token
`

xiongyihui, 2016-02-14:
Which version of Arduino IDE do you use? No error in v1.0.5 and v1.6.4.

ConorOD, 2016-02-15:
Ah, I was using 1.6.6 works fine on 1.6.4

thanks very much

ConorOD, 2016-02-21:
So I fixed the error mentioned above, but when I try to upload it to my board (intel galilleo) i get the same error again, is it possibly a problem with my board?

zapper9595, 2017-06-14:
Hi @ConorOD , not sure if your problem is solved. I'm currently facing the same problem. Do share if you did, thanks.

ConorOD, 2017-06-14:
Hey @zapper9595 I'm not sure what I did, I will try to find the file and see if I can remember

Pillar1989, 2019-08-24:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-24 by Pillar1989