Seeed-Studio/PN532 #42: P2P_Sending limitation

lorant124, 2016-02-15:
Hi,

I would like to ask for help, I m currently using your library of NFC to send messages to android over p2p protocol. I would like to send at least 2000 characters.

Is It possible or it is given by the hardrware PN532, I use ESP8266 with arduino IDE and now I'm able to send 223 byte. I know that the library limites the number of bytes "SNEP packet, the packet should be less than (255 - 2 - 3) bytes". My question is if it was possible to send over this limit?

Thank you very much for your answear and your help,

Lorant

xiongyihui, 2016-02-16:
Packet size is also limited by PN532 command size. Normal packet data size is up to 255. Extended  packet data is up to 65535 but is not supported yet. See [PN532 User Manual 6.2.1.2](http://www.nxp.com/documents/user_manual/141520.pdf)

Another limitation is, at the moment, the library support to exchange only one INFO PDU of P2P communication. changing SNEP::write() to support more INFO PDUs.

diegosilvars, 2016-10-14:
Hello. any news? There is no alternative to get around this limitation?

Pillar1989, 2019-08-24:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-24 by Pillar1989