Seeed-Studio/PN532 #43: Receiving interrupt from NFC shield on D2 pin when target in RF field

mashhashkarma, 2016-02-15:
Hi, this is more of a query abt interrupt driven P2P exchange, than an issue...kindly help
i wish to call LLCP::activate(), on receiving interrupt on D2 pin.(i have connected (soldered) D2 to IRQ on the NFC shield).
If i do the following in setup(),
       attachInterrupt(0,myISR,FALLING);
when will the myISR() be invoked ?

Thanks !

xiongyihui, 2016-02-16:
LLCP::activate() calls PN532::tgInitAsTarget() internally. To use interrupt, PN532::tgInitAsTarget() should be split into two parts: writing command and reading response. The sequence is like:

1. write command
2. wait for interrupt (do something else)
3. read response

If you have IRQ pin connected, PN532_SPI::isReady() cloud be optimized by check IRQ pin instead of checking SPI response.

mashhashkarma, 2016-02-16:
Thanks for the quick reply !!
As I understand,PN532_SPI::isReady() operates on SPI CS/SS pin 10 (PN532_SPI pn532spi(SPI, 10);).
I guess it is used to signal when data needs to be RX or TX over SPI.
I mean, if something needs to be TX'd or RX'd over SPI, pin 10 is being used to signal. 
Is IRQ pin also raised at the same time ? 

I'm trying to understand, in what situations does PN532 raise interrupt on IRQ.
I tried out a sample code 
   **  attachInterrupt(digitalPinToInterrupt(2),myIsr,FALLING);
      nfc.activate();**
This resulted in myIsr() getting called twice. So wondering what events causes PN532 to raise interrupt on IRQ pin.

xiongyihui, 2016-02-16:
IRQ pin raise interrupt when PN532 has data to be read.

In the function nfc.activate(), the sequence is:

1. MCU sends command
2. PN532 raises IRQ to notify data is ready  @ 1
3. MCU reads ACK/NACK
4. PN532 discovers RF field
5. PN532 raises IRQ @ 2 
6. MCU reads response

mashhashkarma, 2016-02-16:
OK.Thanks !
Looks like i have to make lot of modification to make PN532 interface calls non-blocking.
After that i have to figure out how to run both BLE Shield and NFC shield together. SPI SS pins are different but i dont know if SPI driver will keep separate the data transmissions of BLE and NFC shield to/from UNO .

Thank you for your time.

closed 2016-02-29 by xiongyihui