Seeed-Studio/PN532 #44: esp8266

sfranzyshen, 2016-02-16:
Please add support for the esp8266 ... 

Hello!
Found chip PN532
Firmware ver. 1.6
Waiting for an ISO14443A card

never returns anything else ...

xiongyihui, 2016-02-16:
Which interface do you use? It should be easy to change the hal layer (PN532_I2C, PN532_SPI or PN532_HSU) to support esp8266.

sfranzyshen, 2016-02-16:
I am trying to use the SPI ... it finds the board and identifies the firmware ... but then nothing ...

sfranzyshen, 2016-02-16:
OK, I can sort of read a card ... I edited the PN532_debug.h file and uncommented the #define DEBUG line ... when I start up with no card on reader this happens ... and it will not read a card ...
```
Hello!
write:  2
read:   3 32 1 6 7
Found chip PN532
Firmware ver. 1.6
SAMConfig
write:  14 1 14 1
read:   15
Waiting for an ISO14443A Card ...
write:  4A 1 0
write:  4A 1 0
Invalid ACK
write:  4A 1 0
write:  4A 1 0
write:  4A 1 0
```
But when I start up with a card already on the reader ... it seems to read it ... but never finishes or returns the information (only the debug output) to code ...

```
Hello!
write:  2
read:   3 32 1 6 7
Found chip PN532
Firmware ver. 1.6
SAMConfig
write:  14 1 14 1
read:   15
Waiting for an ISO14443A Card ...
write:  4A 1 0
read:   4B 1 1 0 4 8 4 4A 8D 69 4F
ATQA: 0x 4SAK: 0x 8
Found an ISO14443A card
  UID Length: 4 bytes
  UID Value:  4A 8D 69 4F

Seems to be a Mifare Classic card (4 byte UID)
Trying to authenticate block 4 with default KEYA value
write:  40 1 60 4 FF FF FF FF FF FF 4A 8D 69 4F
read:   41 0
Sector 1 (Blocks 4..7) has been authenticated
Trying to read 16 bytes from block  4write:  40 1 30 4
read:   41 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0 0
Reading Block 4:
 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00    ................

write:  4A 1 0
write:  4A 1 0
write:  4A 1 0
Invalid ACK
write:  4A 1 0
Time out when waiting for ACK

```
any help?

xiongyihui, 2016-02-16:
Did you use the example readMifare? It seems the example works as it is.

Maybe you should increase [the delay](https://github.com/Seeed-Studio/PN532/blob/master/PN532/examples/readMifare/readMifare.pde#L84) to read the card again.

closed 2016-02-29 by xiongyihui

sfranzyshen, 2016-03-11:
why was this closed? It is not solved ... is it?

xiongyihui, 2016-03-14:
It seems you've already get it work. You can read the uid of a card.

sfranzyshen, 2016-03-17:
I am bailing on the SPI interface (for now) I have the i2c working flawless with the following changes ...
`
#include <Wire.h>
#include <PN532_I2C.h>
#include <PN532.h>

PN532_I2C pn532i2c(Wire);
PN532 nfc(pn532i2c);

// and in the setup() add 
Wire.setClockStretchLimit(2000);
Wire.begin(0, 2);
nfc.begin();

nfc.setPassiveActivationRetries(0x19);
nfc.SAMConfig();

`
it works well for me with the iso14443a_uid example ... I'm just reading the uid ... so the other stuff is untested ...

AdamMiltonBarker, 2016-06-07:
@sfranzyshen FINALLY :)  Cheers buddy got it working.

AdamMiltonBarker, 2016-06-07:
@sfranzyshen  well it was working, I managed to get it working with tag.getUidString as well but after a couple of scans it crashes