Seeed-Studio/PN532 #50: Continuously read mifarecard with PN532 RFID/NFC Sheild

enkiid, 2016-11-21:
Hi there,

I'm fairly new to programing and using RFID/NFC, I'm currently using a UNO and with Adafruits PN532 RFID/NFC shield. 

I dont know if it's possible to do this but here we go: 

I would like to keep an LED light on _while_ the _same_ UID is being read and turn off as soon as I pick up the card from the shield.

The code I've attached is currently reading the card, checking if its UID is valid, if it is, turn the green LED on, if not turn the red LED on. The problem is if I leave my card on the shield it will restart the process and reread the card (I know i have the return function there but I want to return when the card is not on the shield). 

Thanks in advance!

```
#include <Wire.h>
#include <SPI.h>
#include <Adafruit_PN532.h>

#define PN532_IRQ   (2)
#define PN532_RESET (3)

Adafruit_PN532 nfc(PN532_IRQ, PN532_RESET);

uint8_t success;
uint8_t uid[] = { 0, 0, 0, 0, 0, 0, 0 };  // Buffer to store the returned UID
uint8_t uidLength;                        // Length of the UID (4 or 7 bytes depending on ISO14443A card type)
    

// Add your valid rfid keys here. To find you your key just run sketch; hold your new RFID tag in fron ot the reader; 
// and copy the key from serial output of this sketch.
const uint8_t maxKeyLength = 7;
uint8_t validKeys[][maxKeyLength] = { // Add keys here
                    {0x4E, 0x98, 0x78, 0x73, 0x00, 0x00, 0x00},
                    {0x04, 0x52, 0x38, 0xBA, 0x13, 0x3C, 0x80}
                    };
                    
int keyCount = sizeof validKeys / maxKeyLength;             
boolean valid = false;
const int redLed = 4;
const int greenLed = 5;

void setup()
{  
  Serial.begin(115200);
  Serial.println("Hello!");
  
  pinMode(redLed, OUTPUT);
  pinMode(greenLed, OUTPUT);

  nfc.begin();

  uint32_t versiondata = nfc.getFirmwareVersion();
  if (! versiondata) {
    Serial.print("Didn't find PN53x board");
    while (1); // halt
  }
  // Got ok data, print it out!
  Serial.print("Found chip PN5"); Serial.println((versiondata>>24) & 0xFF, HEX); 
  Serial.print("Firmware ver. "); Serial.print((versiondata>>16) & 0xFF, DEC); 
  Serial.print('.'); Serial.println((versiondata>>8) & 0xFF, DEC);
  
  // configure board to read RFID tags
  nfc.SAMConfig();
  
  Serial.println("Waiting for an ISO14443A Card ...");
}

void loop()
{ 
  digitalWrite(greenLed, LOW);
  digitalWrite(redLed, LOW);
  
  // Wait for an ISO14443A type cards (Mifare, etc.).  When one is found
  // 'uid' will be populated with the UID, and uidLength will indicate
  // if the uid is 4 bytes (Mifare Classic) or 7 bytes (Mifare Ultralight)
  boolean success;
  success = nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, &uid[0], &uidLength);
  
 if (success) {
    // Display some basic information about the card
    Serial.println("Found an ISO14443A card");
    Serial.print("  UID Length: ");Serial.print(uidLength, DEC);Serial.println(" bytes");
    Serial.print("  UID Value: ");
    nfc.PrintHex(uid, uidLength);

   
    // Compare this key to the valid once registered here in sketch 
    for (int i=0; i < keyCount && !valid; i++) {
      for (int j=0; i < uidLength && !valid; j++) {
        if (uid[j] != validKeys[i][j]) {
          break;
        }
        if (j == uidLength-1) {
          valid = true;
        }
      }
    }
    if (valid) {
      valid_card() ;       
    }
    
    else {
      invalid_card() ;   
    }
  }
}

void valid_card()
{
  Serial.println("Valid card!");
  digitalWrite(redLed, LOW);
  digitalWrite(greenLed, HIGH);
  delay(2000);
  return;
}

void invalid_card()
{
  Serial.println("Invalid card!");
  digitalWrite(greenLed, LOW);
  digitalWrite(redLed, HIGH);
  delay(2000);
  return;
}
```

xiongyihui, 2016-11-22:
It seems you used the Adafruit PN532 NFC library, not this one.

There is a timeout parameter for `bool readPassiveTargetID(uint8_t cardbaudrate, uint8_t *uid, uint8_t *uidLength, uint16_t timeout = 1000, bool inlist = false);`, it may solve the issue.

https://github.com/Seeed-Studio/PN532/blob/master/PN532/PN532.cpp#L289

enkiid, 2016-11-22:
Thanks for the reply, sorry for my stupidity, will post in Adafruits forum! 

Dont really know how I would implement that into my code, I totally understand if you dont want to help haha!

xiongyihui, 2016-11-22:
You didn't get that.

I mentioned that you used the adafruit pn532 library cause I was not sure there is a timeout parameter of the function in adafruit's library. In this library, there is a timeout parameter.

Just add a timeout to `nfc.readPassiveTargetID` in your code, for example

```
success = nfc.readPassiveTargetID(PN532_MIFARE_ISO14443A, &uid[0], &uidLength, 3000);
```

closed 2016-11-22 by xiongyihui