Seeed-Studio/PN532 #54: Add support for SoftwareSerial (PN532 control via UART)

basil-ionin, 2017-03-17:
Hello,

I have Arduino UNO, so I can handle only one hardware serial, but I use it for transfering data between Arduino and my PC via USB.

Why cant I re-define GPIO pins for UART, because it would be easy to work with my RFID PN532. If it's impossible, can you explain why or how can I solve this problem?

Thank you

P. S. Sorry for my English, it's not my native language.

xiongyihui, 2017-07-26:
SoftwareSerial is not very reliable. Not recommend to use it.
You may need to change [HardwareSerial](https://github.com/Seeed-Studio/PN532/blob/master/PN532_HSU/PN532_HSU.h#L14) to SoftwareSerial if you want to use SoftwareSerial

francesco98, 2017-11-25:
Hello, I have had recently the same problem. I'm fixing it changing HardwareSerial to SoftwareSerial, but you have to do others changes in code because you will receive some errors in some lines in which is write(int) function in ambigous way. For exemple _Serial.write(0) is ambigous and you must change it in _Serial.write((byte)0). 
Now, my code is finished and correctly verified by Arduino IDE, but I can't test it because I haven't the PN532 shield yet.

kempy007, 2018-02-02:
Can you share your changes??

oscherler, 2018-07-10:
I modified elechouse’s PN532_HSU to use SoftwareSerial in a new library called PN532_SWHSU. 

https://gist.github.com/oscherler/2df3f13140cd55f6a8371a9b40331355

Instead of

```
#include <PN532_HSU.h>
#include <PN532.h>
      
PN532_HSU pn532hsu(Serial1);
PN532 nfc(pn532hsu);
```

use

```
#include <SoftwareSerial.h>
#include <PN532_SWHSU.h>
#include <PN532.h>

SoftwareSerial SWSerial( 10, 11 ); // RX, TX

PN532_SWHSU pn532swhsu( SWSerial );
PN532 nfc( pn532swhsu );
```

Pillar1989, 2019-08-25:
refer to https://github.com/Seeed-Studio/PN532/commit/4828348c777cc3cc6208d08667d14a3aa81cab67

closed 2019-08-25 by Pillar1989