Seeed-Studio/PN532 #58: No response when using Xadow NFC

Pida23, 2017-07-26:
I connect Xadow NFC with the Xadow Main Board but when I uploaded the example code of WriteTag, no response at all. The code can be compiled and uploaded nicely but at the end nothing is happening. I already check the connection and it's all good. Any helps?

Thank you in advance.

xiongyihui, 2017-07-26:
Which interface do you use? SPI or I2C

Pida23, 2017-07-26:
I'm not sure which interface that I'm using.. but i tried both, I2C and SPI. if you see at the back of the Xadow NFC module, there is an option of SPI, I2C and UART. My computer only detect the usb when I set the I1 to High and I0 to Low.

Pida23, 2017-07-26:
[image]

xiongyihui, 2017-07-26:
By default, the Xadow NFC use I2C interface. If it doesn't work, you can try to use  the uart interface. cut the wire between SDA pad and the center pad, SCL pad and another center pad, and then change the code from

```
#include <SPI.h>
#include <PN532_SPI.h>
#include "PN532.h"

PN532_SPI pn532spi(SPI, 10);
PN532 nfc(pn532spi);
```
to
```

#include <PN532_HSU.h>
#include "PN532.h"

PN532_HSU pn532hsu(Serial1);
PN532 nfc(pn532hsu);
```

Pida23, 2017-07-27:
As you said the default is I2C means if I don't do anything to the board so it will operate using I2C interface, am I right? But in that condition, my computer can't detect the main board until I put solder on I1 and I0. And one more thing, refer to my photo above, what interface I'm currently in?
Thank you.

Licsber, 2017-07-27:
It seems that you are in SPI mode . Why not use UART mode , I think it is more easy to control.

Pida23, 2017-08-02:
As mentioned above, to use UART i need to cut the wire between SDA pad and center pad. Then also cut the wire between SCL pad and another pad. Am i right? And how about I1 and I0 that i soldered before? should I removed the solder or just leave it there?

Thank you.

Pida23, 2017-08-07:
Hello guys?

Pillar1989, 2019-08-25:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-25 by Pillar1989