Seeed-Studio/PN532 #69: Problem requesting information from PN532 in NFC Type4 card emulation mode with Samung Galaxy S6

josebagarciamiranda, 2018-06-07:
Hello, everyone:

I am having some troubles using the PN532 shield on an arduino with my Samung Galaxy S6. I am trying to emulate an NFC Type 4 tag with _emulate_tag_ndef.ino_ example for the arduino. The problem is that, meanwhile I have no problems with a Huawei G6 Ascend phone to get all data through the nfc-tools app, I do not get nothing in my Galaxy S6 almost every time. I get information from the PN532 once per 20 tries.

I  have been looking for more information and all that I get is that the Galaxy S6 has a S3FWRN5P NFC chip that does not support MIFARE Classic tags. However, according to SAK and ATQA values in _emulatetag.cpp_, the PN532 tries to emulate a MIFARE PLUS tag, which is supported as far as I know. I even try to change  SAK and ATQA values in order to emulate other tags but I still have the same problems in the S6.

Does anyone know if there is an issue between the S3FWRN5P NFC chip from the S6 and the PN532? I tried to go deep into the _emulatetag.cpp_ file but I do not know what more I can change. Furthermore, I do not even know if both NFC chips are compatible and I must purchase another NFC board.

Thank you very much in advance.

coelner, 2018-06-08:
See here: https://www.shopnfc.it/en/content/7-nfc-compatibility
Mifare Plus is similiar to the Classic. I'm not sure, but usually Samsung is not very kind in following standards. Additionally there is a set of Broadcom chips which are ignoring Mifare at all/partially. 

 Try a least this example: https://github.com/Seeed-Studio/PN532/blob/master/PN532/examples/p2p_with_ndef_library/p2p_with_ndef_library.ino
It should work for Android 5+

josebagarciamiranda, 2018-06-08:
Thank you very much for the answer, coelner. I checked the _p2p_with_ndef_library.ino_ previously. I was more interesed in emulating a tag and that is why I continued with _emulatetag.cpp_ example.

However, regarding the compatibility issues between different NFC chips, as you have shown in the link from shopnfc, and the Android fragmentation, where almost the 80% of the Android devices go from KitKat onwards,  maybe I shuold make a try performing a p2p communication and not a tag emulation.

Thanks!

Pillar1989, 2019-08-25:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-25 by Pillar1989