Seeed-Studio/PN532 #75: MCU Reset when read a NFC Forum Type 2 tag twice!

ThomasW69, 2019-02-27:
I use this library on an ESP32. The communication with the reader works fine. I also can read the information of the tag. With Mifare classic everything is OK. The tag will be read everytime the read function is called in the loop().
When I use a NFC Forum Type 2 tag then the loop() runs only once. If the loop() runs a second time my MCU is resetting with an error message. 
This is the only code I execute.
`void loop() {
  if (nfc.tagPresent())
      {
          tag = nfc.read();
          Serial.println(tag.getTagType());
      }
      delay(3000);
 }`  
In the terminal window I get the following messages 

` Found chip PN532
Firmware ver. 1.6
NFC Forum Type 2
assertion "heap != NULL && "free() target pointer is outside heap areas"" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/heap/heap_caps.c", line 268, function: heap_caps_free
abort() was called at PC 0x400d4f03 on core 1`

ThomasW69, 2019-02-27:
maybe this could help to finde the problem. 
[https://github.com/espressif/arduino-esp32/issues/1101](url)

Pillar1989, 2019-08-25:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-25 by Pillar1989