Seeed-Studio/PN532 #76: SNEP code -4

sokol07, 2019-04-09:
Hi,
I'm using PN532 with ESP8266 and Xiaomi Mi8, I'm sending text data as NDEF message through NFC using "Legic NFC P2P Example" app, so I'm sending SNEP messages through Android Beam.
I've got code like this:
```
#include "snep.h"
#include "NdefMessage.h"
PN532_SPI pn532spi(SPI, 0);
SNEP nfc(pn532spi);
uint8_t ndefBuf[128];
uint16_t timeout = 100000;

void setup()
{
    Serial.begin(115200);
    Serial.println("start");
    delay(100);
}

void loop()
{
  Serial.println("Get a message from Android");
    int msgSize = nfc.read(ndefBuf, sizeof(ndefBuf),timeout);
    Serial.println(msgSize);
    if (msgSize > 0) {
        NdefMessage msg  = NdefMessage(ndefBuf, msgSize);
        NdefRecord record = msg.getRecord(0);

        int payloadLength = record.getPayloadLength();
        byte payload[payloadLength];
        record.getPayload(payload);        
        
        // The TNF and Type are used to determine how your application processes the payload
        // There's no generic processing for the payload, it's returned as a byte[]
        int startChar = 0;        
        if (record.getTnf() == TNF_WELL_KNOWN && record.getType() == "T") { // text message
          // skip the language code
          startChar = payload[0] + 1;
        } else if (record.getTnf() == TNF_WELL_KNOWN && record.getType() == "U") { // URI
          // skip the url prefix (future versions should decode)
          startChar = 1;
        }
                          
        // Force the data into a String (might fail for some content)
        // Real code should use smarter processing
        String payloadAsString = "";
        for (int c = startChar; c < payloadLength; c++) {
          payloadAsString += (char)payload[c];
        }
            
        Serial.println(payloadAsString);
        Serial.println("\nSuccess");
            } else {
        Serial.println("failed");
    }
    delay(500);
}
```
In further part of the code it displays the message on a screen, but this isn't important in this issue.
My problem is that this code works as long as I'm sending message with these parameters:
TNF: Well known
Type: text/plain
Payload is up to 105 bytes. If it's shorter - no problem. If it's longer... Well, that's where the issue begins.
If I have longer payload, the SNEP::read function returns "-4", which I think means that the message is longer than 255 bytes.
Is there any walkaround, letting me send longer data sets with Android Beam?

Pillar1989, 2019-08-25:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-25 by Pillar1989

BarisUygun, 2020-11-18:
Have you solved this issue? I am trying to send a string message from an android device to pn532 which is connected to a nodemcu. Nodemcu can not get the message it only gets the rfid of the android device. How do i fix this? I am using the I2C mode.