Seeed-Studio/PN532 #78: limited set UID byte on Emulate Tag

ikhdarisnan, 2019-04-20:
i just create new project for my undergraduated final task. i want to do a relay attack and using tag emulation on my PN532, but what i got the UID can't set perfectly  what i want. The first UID byte is masked by the library using 0x08. so i got stuck by this problem. The example code say "the UID must be 3 bytes" and when i edited the array of UID for 4 bytes its still masked with the "0x08" on first byte.

Someone can tell me the solution for this problem? i use esp8266 + PN532 V3 module with SPI interface

dimonDDL, 2019-05-10:
Isn't this behavior baked in to the chip for protection?

ikhdarisnan, 2019-06-30:
i don't know right, but i found a NFC Security researcher that can manipulate it and make it right without the "0x08" thing

Kelssmith, 2019-07-01:
I keep getting Didn't find PN53x board

do you know why this happens?

*Kel Smith *
NED, LISTME
07918905702  | ks@listmeapp.co | www.listmeapp.co | London

This communication contains information which is confidential, which may be
privileged, and which is for the exclusive use of the intended
recipient(s). If you are not an intended recipient please note that any
distribution, disclosure, use or copying of any part of this communication
is strictly prohibited. If you have received this communication in error
please notify us by return email or by telephone and delete this
communication and any copies of it. Statements or opinions may be expressed
in this communication that are personal to the sender and do not
necessarily represent the views of  LIST ME LTD or any member of LIST
ME LTD warrants
that this email is free from error, viruses, malware, data-damaging
material or other defects, or is compatible with your equipment or fit for
any purpose.

Unless expressly stated otherwise, no Affiliated Association and no member
of LIST ME LTD shall be bound by any contract or obligation purported to be
created by this communication.

This communication has originated from the systems network of LIST ME LTD
 who may monitor, intercept and block emails addressed to its users or take
any other action in accordance with its email use policy. LIST ME LTD is
registered in England and Wales, with its registered office at LISTME
LTD, Suite
20, Neals Corner, Middlesex, TW3 3HJ  www.listmeapp.co

ikhdarisnan, 2019-07-13:
maybe you have a problem with the cable.
make sure you choose the right interfaces (SPI, I2C, UART). To emulate a tag you should use SPI interface and before uploading the code, unplug the SS pin first. then after the code compiled and while the code starts to upload, plug the SS pin.

ikhdarisnan, 2019-07-13:
> I keep getting Didn't find PN53x board do you know why this happens? *Kel Smith * NED, LISTME 07918905702 | ks@listmeapp.co | [www.listmeapp.co](http://www.listmeapp.co) | London This communication [...]

maybe you have a problem with the cable.
make sure you choose the right interfaces (SPI, I2C, UART). To emulate a tag you should use SPI interface and before uploading the code, unplug the SS pin first. then after the code compiled and while the code starts to upload, plug the SS pin.

ikhdarisnan, 2019-07-13:
Based on articles, paper, and PN532 datasheet, I didn't have any solutions. The main purposes for limited to set UID by full custom UID are for security. Hopefully, there are some solutions in the future.

closed 2019-07-13 by ikhdarisnan