Seeed-Studio/PN532 #81: Can't connect with Java Card

il3ven, 2019-07-06:
I am trying to communicate with a Java Card using PN532 and Arduino.
When I call `inListPassiveTarget()` it returns false whenever I tap my card.

I was using [ACR122u ](https://www.acs.com.hk/en/products/3/acr122u-usb-nfc-reader/)before which is based on PN532 with [PyAPDU ](https://javacardos.com/tools)Tools and it worked fine.

Pillar1989, 2019-08-25:
I'll close the question and open it again if you need to discuss it later

closed 2019-08-25 by Pillar1989

il3ven, 2019-08-25:
### Solution
After many hit and trials this is what worked for me.
1. Changed I2C to SPI
2. Used [Adafruit's Library](https://github.com/adafruit/Adafruit-PN532)

If the above doesn't work for someone. Try changing your PN532 chip. There are many Chinese clones that sometimes don't work properly.