Seeed-Studio/PN532 #87: emulate_tag UID

kjubow, 2019-09-08:
when i run [emulate_tag_ndef](https://github.com/Seeed-Studio/PN532/blob/arduino/PN532/examples/emulate_tag_ndef/emulate_tag_ndef.ino) sketch and read the PN532 with phone the UID is 00:00:00:00:00:00:00:00
and not 12:34:56 as it's written int the code. however i achieved changing UID by editing the NFCID2 (FeliCa Params) in uint8_t command[] in [emulatetag.cpp](https://github.com/Seeed-Studio/PN532/blob/arduino/PN532/emulatetag.cpp#L90). so i don't know if it's on purpose or not. 

also i want to know if it's possible to change the lenght of UID. so it's not 3 or 10 bytes long but for example 4 bytes long.

thanks

Pillar1989, 2019-09-09:
@jakubkrnac What hardware do you use?  Maybe you can try our library and our supporting products for testing.

jakubkrnac, 2019-09-09:
i am using chinese PN532 module witch is probably FeLica so that's why it uses the NFCID2 and not NFCID1. so let's forget about that. 

i want to know if it's possible to change the lenght of the UID. 

thanks

Pillar1989, 2019-09-09:
Is this the hardware?
[image]

jakubkrnac, 2019-09-09:
no it's not i said that is some chinese module. can you please answer if it's possible to change the lenght of UID??

Pillar1989, 2019-09-10:
Different hardware in the software driver is still very different, I suggest you still where to buy, you go to him for help, this software is not a universal software.

closed 2019-09-10 by Pillar1989