Seeed-Studio/PN532 #88: SPI card emulation is not working

Adrianotiger, 2019-09-16:
I have the elechouse v3 module (or probably a clone, since I don't know anymore when I purchased it and can't find out if it is a clone or not).

After setting up everything like in the example for the emulate_tag_ndef.h, I have this loop:
```
// the loop function runs over and over again until power down or reset
void loop()
{
  Serial.println("Begin emulation.");
  if (!nfc.emulate(10000))timeout 10 seconds
    Serial.println("timed out");
  else
    Serial.println("mobile device found!");

  if (nfc.writeOccured()) {
    Serial.println("\nWrite occured !");
    uint8_t *tag_buf;
    uint16_t length;

    nfc.getContent(&tag_buf, &length);
    NdefMessage msg = NdefMessage(tag_buf, length);
    msg.print();
  }
  delay(1000);
}
```

It should emulate an NFC tag with the seeeds studio tag. Unfortunately, it doesn't send the url and gives the wrong data to the phone.
[image]

Enabling the debug on the library, this is what I get once the phone will answer:

> Begin emulation.
> write:  8C 05 04 00 12 34 56 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
> read:   8D 08 E0 80
> write:  86
> read:   87 13
> status is not ok
> tgGetData failed!
> write:  52 00
> read:   53 00
> mobile device found!

Shouldn't it be able to emulate a mifare? I just don't know if it is something with the library or if it is something with the module.

Adrianotiger, 2019-09-17:
After a lot of googling and changes in the library, I found some errors and found some info about it.
As said from a lot of users, the NXP documentation isn't really good and so a lot of info are missing. But the library structure is good and works. Just some info about it:

Starting with this tgInitAsTarget:
```
  uint8_t command[] = {
      PN532_COMMAND_TGINITASTARGET,
      5, // MODE: PICC only, Passive only

      0x04, 0x00,       // SENS_RES
      0x00, 0x00, 0x00, // NFCID1
      0x20,             // SEL_RES

      0, 0, 0, 0, 0, 0, 0, 0,
      0, 0, 0, 0, 0, 0, 0, 0, // FeliCaParams
      0, 0,

      0, 0, 0, 0, 0, 0, 0, 0, 0, 0, // NFCID3t

      0, // length of general bytes
      0  // length of historical bytes
  };
```
On the last commit you changed it and now it is wrong. The PN532 is emulating a FeliCa card and you have to set it up correctly.
So, you have to write it in this way:

```
  uint8_t command[] = {
      PN532_COMMAND_TGINITASTARGET,
      0x05,                  // MODE: 0x04 = PICC only, 0x01 = Passive only (0x02 = DEP only)

      // MIFARE PARAMS
      0x04, 0x00,         // SENS_RES (seeeds studio set it to 0x04, nxp to 0x08)
      0x00, 0x00, 0x00,   // NFCID1t	(is set over sketch with setUID())
      0x20,               // SEL_RES    (0x20=Mifare DelFire, 0x60=custom)

      // FELICA PARAMS
      0x01, 0xFE,         // NFCID2t (8 bytes) https://github.com/adafruit/Adafruit-PN532/blob/master/Adafruit_PN532.cpp FeliCa NEEDS TO BEGIN WITH 0x01 0xFE!
      0x05, 0x01, 0x86,
      0x04, 0x02, 0x02,
      0x03, 0x00,         // PAD (8 bytes)
      0x4B, 0x02, 0x4F, 
      0x49, 0x8A, 0x00,   
      0xFF, 0xFF,         // System code (2 bytes)
      
      0x01, 0x01, 0x66,   // NFCID3t (10 bytes)
      0x6D, 0x01, 0x01, 0x10,
      0x02, 0x00, 0x00,

	  0x00, // length of general bytes
      0x00  // length of historical bytes
  };
```
Some more info on https://www.nxp.com/docs/en/nxp/application-notes/AN133910.pdf page 51 (how to fill TgInitAsTarget parameters?). I am not sure about it, but after setting it correctly, my phone begun to read the NDEF Tags.

This is not all. An important point is the first tgGetData call.
You probably get this one:
> write:  86
> read:   87 13
if you have a cloned shield or if the NFC-shield is not correctly positioned on your smartphone... Just continue to reading or polling will help you to get a right answer.

Once you get another answer, this one could be:
> write:  86
> read:   87 29
This is a good answer! The smartphone replied with "target has been released by its initiator". It means that the message was arrived to destination and the smartphone want to read more (https://stackoverflow.com/questions/24777810/the-pn532-configured-as-target-has-been-released-by-its-initiator). So, if you get this 0x29 answer, just init the PN532 again and the smartphone will read the Tag correctly. You have to modify the `pn532.cpp - tgGetData()` to parse the 0x29 command and you have to modify the `emulatetag.cpp - emulate()` to call `if (pn532.tgInitAsTarget(command, sizeof(command), 1000) == 1) continue;` if the result of the status was 0x29 and it is the first command read inside the loop.

I am still playing with this module, so maybe this is not the best solution. But it works now! If you want I can create a fork and commit my changes... But as said, it is probably not the perfect solution... Just a working solution.

Pillar1989, 2019-09-26:
@Adrianotiger 
Sorry, I was too busy these days and didn't see your issue. According to your analysis, you are an excellent questioner. Hopefully someone will come up with the perfect solution.

Adrianotiger, 2019-09-26:
Hi Pillar,
I was able to implement it. Just a note:
In the last commit, you overwrote the right FeliCa parameters:
https://github.com/Seeed-Studio/PN532/commit/ac34962054dfa9786adde3e0f10ae227f9b0ee90#diff-d96d2abf3208c281c9630c9602a4ac8fL73

As the module can just emulate a FeliCa, it is important the the first 2 bytes are 0x01 0xFE. Without this, some Android phones could detect a manipulation or fake card and doesn't reply.

The NDEF library is also really buggy, if you try to add 2 NDEF messages, your Arduino will crash. There are some pull-request to solve all the memory leaks bugs but was never commited in the main library.

Pillar1989, 2020-01-15:
I'll close it first, and if you need to discuss it later, you can open it again

closed 2020-01-15 by Pillar1989

sumspace, 2020-02-06:
can you please help me how can i confiure the emulated tag as ISO1443-3A, the given configuration emulate as Felica type.
 i am trying this so that it can work with ISO app. as what ever i could try is that android is able to read both felica and 14443-3a type but using IOS i am not able to read the emulated tag. not even the real tag. 
Just one more thing to confirm , what ever i could found is that IOS does n't support HCE mode , right ? that is the reason i need to emulate PN532 as ISO14443A.

freeridre, 2021-04-26:
Hi there!
I also try to emulate Pn532 as a tag, and I get exactly what @Adrianotiger posted previously in nfc tools app. I was not able to imolement to send ndef message to android.
Cloud somebody help me?
Thank so much!

cino1001, 2021-05-15:
Hi everyone 
I get the answer 0x29, someone can help me on how to init again the PN532

jlkalberer, 2022-05-30:
@Adrianotiger 

You said you figured out a solution to fix things so emulation works on the PN532?
Can you share your answer?

Adrianotiger, 2022-06-08:
> @Adrianotiger
> 
> You said you figured out a solution to fix things so emulation works on the PN532? Can you share your answer?

I'm sorry... It was a lot of time ago. I never made a fork of the project. But with the code provided in this issue, I was able to emulate a card.

You must debug, see what you get as reply from the module. The module can't start as card emulator if I can remember it. It always restarts as emulator once you put your phone near the module. So, the first answer is 87 29.

atg1966, 2022-08-22:
> I'm sorry... It was a lot of time ago. I never made a fork of the project. But with the code provided in this issue, I was able to emulate a card.
>
> You must debug, see what you get as reply from the [...]

@Adrianotiger 

Did you manage to emulate a Mifare Classic 14443-3A card ? Wondering because you said once "As the module can just emulate a FeliCa...". Thank you !

c4r3, 2023-08-16:
Hi @Adrianotiger, i know has been a long time but i have the same issue: i got the same screenshot with NFC Tools when with the first try of card emulation with PN532. Then i change the settings of "tgInitAsTarget" as you suggested but still i'm not capable of a 100% reading. 
Do you change anything else? Thanks in advance.

Adrianotiger, 2023-08-16:
> Hi @Adrianotiger, i know has been a long time but i have the same issue: i got the same screenshot with NFC Tools when with the first try of card emulation with PN532. Then i change the settings of [...]

I can't remember :P
I think, I begun to make multiple reads. So if I got an incorrect answer, I just made the request again.
The module works also better with a high voltage. They write from 2.7 to 5.5V. With 4.5V my module begun to give strange answers... I think you should try a good power supply with 1A and 5V. To have a stable module.

c4r3, 2023-08-16:
@Adrianotiger i'm managing the module with an ESP8862 board and i'm using 3.3 voltage available. Actually you are right, at first i need 5V to have a stable behavior of the module itself.
On the other hand if the problem still persists i have to check step by step the communication i think...

Dermiste, 2024-02-20:
Hi @Adrianotiger, I guess it's my turn to step on this issue and try to make things work. It's been a while but do you possibly have code pieces to share regarding this?

antonsimola, 2024-02-24:
Hi guys @Dermiste  @Adrianotiger. I managed to parse horrendous fix together, based on Adrianos explanation (forgive me, I have 10h Arduino experience).

emulatetag.cpp, emulate function (whole function, but only changes in the command array and in the runloop retry stuff)

```cpp
bool EmulateTag::emulate(const uint16_t tgInitAsTargetTimeout)
{

    //https://github.com/Seeed-Studio/PN532/issues/88
    uint8_t command[] = {
      PN532_COMMAND_TGINITASTARGET,
      0x05,                  // MODE: 0x04 = PICC only, 0x01 = Passive only (0x02 = DEP only)

      // MIFARE PARAMS
      0x04, 0x00,         // SENS_RES (seeeds studio set it to 0x04, nxp to 0x08)
      0x00, 0x00, 0x00,   // NFCID1t	(is set over sketch with setUID())
      0x20,               // SEL_RES    (0x20=Mifare DelFire, 0x60=custom)

      // FELICA PARAMS
      0x01, 0xFE,         // NFCID2t (8 bytes) https://github.com/adafruit/Adafruit-PN532/blob/master/Adafruit_PN532.cpp FeliCa NEEDS TO BEGIN WITH 0x01 0xFE!
      0x05, 0x01, 0x86,
      0x04, 0x02, 0x02,
      0x03, 0x00,         // PAD (8 bytes)
      0x4B, 0x02, 0x4F, 
      0x49, 0x8A, 0x00,   
      0xFF, 0xFF,         // System code (2 bytes)
      
      0x01, 0x01, 0x66,   // NFCID3t (10 bytes)
      0x6D, 0x01, 0x01, 0x10,
      0x02, 0x00, 0x00,

	  0x00, // length of general bytes
      0x00  // length of historical bytes
  };
  if (uidPtr != 0)
  { // if uid is set copy 3 bytes to nfcid1
    memcpy(command + 4, uidPtr, 3);
  }

  if (1 != pn532.tgInitAsTarget(command, sizeof(command), tgInitAsTargetTimeout))
  {
    DMSG("tgInitAsTarget failed or timed out!");
    return false;
  }

  uint8_t compatibility_container[] = {
      0, 0x0F,
      0x20,
      0, 0x54,
      0, 0xFF,
      0x04,                                                        // T
      0x06,                                                        // L
      0xE1, 0x04,                                                  // File identifier
      ((NDEF_MAX_LENGTH & 0xFF00) >> 8), (NDEF_MAX_LENGTH & 0xFF), // maximum NDEF file size
      0x00,                                                        // read access 0x0 = granted
      0x00                                                         // write access 0x0 = granted | 0xFF = deny
  };

  if (tagWriteable == false)
  {
    compatibility_container[14] = 0xFF;
  }

  tagWrittenByInitiator = false;

  uint8_t rwbuf[128];
  uint8_t sendlen;
  int16_t status;
  tag_file currentFile = NONE;
  uint16_t cc_size = sizeof(compatibility_container);
[... 229 lines cut ...]
5V <-> VCC

use PIN 11 as the SS pin:

so in the example, swap the argument  SS ->  PN532_SPI pn532spi(SPI, 11);

```cpp 
#include "emulatetag.h"
#include "NdefMessage.h"

#if 1
  #include <SPI.h>
  #include <PN532_SPI.h>
  #include "PN532.h"

  PN532_SPI pn532spi(SPI, 11);
  EmulateTag nfc(pn532spi);
#elif 0
  #include <PN532_HSU.h>
  #include <PN532.h>
      
  PN532_HSU pn532hsu(Serial1);
  EmulateTag nfc(pn532hsu);
#endif
```

Dermiste, 2024-02-26:
Ok so in the end it worked by brute forcing the results if I understand correctly ?

The expected results are a tag poping up in apps like NFC Tools, right?

jlkalberer, 2024-02-27:
@Dermiste - from what I understand, the NFC protocol has more to the handshake than what is implemented in this library. What's happening is that during the handshake an issue happens (like bad signal) and the protocol tries to reconnect/retry (sending the `0x29`) part of the handshake. 

This workaround above is a brute force to just start the whole handshake over instead of using what's built into the protocol. If this can work reliably enough, I'll totally use the brute force approach :D

---

In theory this should make NFC tools work. I haven't had time to actually test this out and am waiting for someone else to verify the change ;)