Seeed-Studio/PN532 #92: Emulating and reading at same time

giorgio80, 2020-01-25:
Hello , 

For my project I would need both to emulate and reading card .

It would be possible to do that as same time ?

My idea is to read card for half seco d and emulate card for same amount of time . To make the process transparent to user . 

Is this possible or I would need 2 modules?

Pillar1989, 2020-02-07:
You definitely need two modules.

cyberknight01, 2021-08-20:
> Hello ,
> 
> For my project I would need both to emulate and reading card .
> 
> It would be possible to do that as same time ?
> 
> My idea is to read card for half seco d and emulate card for same amount of time . To make the process transparent to user .
> 
> Is this possible or I would need 2 modules?

HI, do you finsh it ? i am doing the same thing, but i cannot emulate

DavidDHansson, 2024-03-06:
I am in the same situation.

Cincinnatu, 2024-10-16:
> Hello ,
> 
> For my project I would need both to emulate and reading card .
> 
> It would be possible to do that as same time ?
> 
> My idea is to read card for half seco d and emulate card for same amount of time . To make the process transparent to user .
> 
> Is this possible or I would need 2 modules?

This solution sounds feasible. Has the issue been resolved now?

DavidTiimo, 2024-10-16:
> This solution sounds feasible. Has the issue been resolved now?

I tried this, and it dosen't work optimally. It gives a bad user experience. I can share my code if you want to try it.

Cincinnatu, 2024-10-16:
> > > 您好，对于我的项目，我需要模拟和读取卡。可以同时这样做吗？我的想法是读取卡片半秒并模拟相同时间的卡片。使进程对用户透明。这是否可能，或者我需要 2 个模块？
> > 
> > 
> > 这个解决方案听起来是可行的。问题现在解决了吗？
> 
> 我试过这个，但它并没有发挥最佳效果。它给人一种糟糕的用户体验。如果您想尝试，我可以分享我的代码。

Okay

closed 2024-11-04 by Lesords