Seeed-Studio/PN532 #93: Incorrect path for .h and .cpp files

tigoe, 2020-02-25:
All the .h and .cpp files have file paths that cause them to fail on compile in the Arduino IDE 1.8.12. To fix this, change all paths that are like this:

````
#include "/PN532/PN532/PN532_debug.h"
````

To this:
````
#include "PN532_debug.h"
````

There are approx. 36 changes across all the files of the various libraries.

Tested by compiling for the Arduino Uno on MacOS and Windows

spiritonline, 2020-05-03:
Thanks this solved my issues!

morgan-wild, 2020-06-01:
+1

I'm annoyed with PN532_SPI.h and #include "PN532/PN532/PN532Interface.h"

:)

Johannf78, 2020-06-01:
Why?

Just remove the paths in front of the filenames...

Pillar1989, 2020-06-02:
https://github.com/Seeed-Studio/PN532/issues/95

closed 2020-06-02 by Pillar1989

morgan-wild, 2020-06-04:
> Why? Just remove the paths in front of the filenames...

I think it is not a good practise to edit the content of a library (directly linked to github by PlatformIO).
Maybe I'm wrong but if an update is done, my project breaks and I need to fix once again theses bad pathes.