Seeed-Studio/PN532 #95: Linux paths incompatible with windows paths

Johannf78, 2020-04-24:
Libraries not found on all windows machines.
For example:
#include <PN532/PN532_SPI/PN532_SPI.h>

Forward slashes do not work in windows machines.
Please change to just one directory level deep and only.
#include <PN532_SPI.h>

Johannf78, 2020-05-14:
Still having the same problem.
Code:
#include "PN532_SPI.h"

Error message:
ResolveLibrary(PN532/PN532/PN532Interface.h)
In file included from D:\OneDrive\Ardruino\NFC Reader\Seeed\Example Code\2020-05-14\2020-05-14.ino:1:0:
  -> candidates: []

C:\Program Files (x86)\Arduino\libraries\PN532_SPI/PN532_SPI.h:6:10: fatal error: PN532/PN532/PN532Interface.h: No such file or directory

 #include "PN532/PN532/PN532Interface.h"

          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~

compilation terminated.

Johannf78, 2020-05-14:
I managed to fix it.
One has to open each and every file in the library and remove the path PN532/PN532_SPI/
So just the file name included must remain.
Open all .cpp and all .h files in the library and edit them....

net234, 2020-05-17:
Fixed :)

closed 2020-06-02 by Pillar1989