Seeed-Studio/PN532 #96: Message chaining for messages > 262 bytes

vmfox, 2020-05-05:
Hi,

I'm using the PN532 to communicate with an android phone as:

Arduino Mega 2560 -> PN532 -> Android

After setting everything up and testing the basic data transfer I got to a point when I need to send data larger than 262 bytes supported by the inDataExchange.

Is there currently a function or a method/approach to send large messages i.e. as described in the chaining mechanism 7.4.5 DEP chaining mechanism - [here](https://www.nxp.com/docs/en/user-guide/141520.pdf)?

I had a look through the library and couldn't find anything relevant and as I don't really have any spare time to append the library, would appreciate some guidance/code drafts along these lines.

Thanks.

vmfox, 2020-05-07:
Potentially relevant [stackoverflow question](https://stackoverflow.com/questions/47820902/cannot-send-large-apdu-commands-with-libnfc-using-nfc-initiator-transceive-bytes)

Cincinnatu, 2024-10-16:
Has your problem been resolved?

closed 2024-11-04 by Lesords