Seeed-Studio/PN532 #99: Call of overloaded 'write(int)' is ambiguous

kordeiro, 2020-06-04:
I get the following error: (PN532_HSU\PN532_HSU.cpp:21:21: error: call of overloaded 'write(int)' is ambiguous) when trying to use the PN532_HSU lib.

Ariel-International, 2021-02-04:
Solved!
In library file PN5323_HSU.cpp one must specify data type every time an integer is used, pecause it could be a pointer, which is not.
Example from line 16 on:
`void PN532_HSU::wakeup()
{
    _serial->write(0x55);                          //_serial->write((byte)0x55);
    _serial->write(0x55);                          //_serial->write((byte)0x55);
    _serial->write(uint8_t(0x00));            //_serial->write((byte)0x00);
    _serial->write(uint8_t(0x00));            //_serial->write((byte)0x00);
    _serial->write(uint8_t(0x00));            //_serial->write((byte)0x00);
`

Line 52 is correct:                            _serial->write(uint8_t(PN532_PREAMBLE));
But line 60 again is not:                  _serial->write(PN532_HOSTTOPN532);

This error shows up only in some platforms, ie. SAMD.
It's a different compiler with different requirements.

kordeiro, 2021-02-04:
Thanks so much! I will give it a try this evening 👍

Cincinnatu, 2024-10-16:
> 非常感谢！今晚👍我会试一试

Hello Is the issue resolved now?

closed 2024-11-04 by Lesords