Seeed-Studio/PN532 #100: Apple Pay activation

giorgio80, 2020-06-12:
Dear, 
need to read raid card only for my application. 

When attaching an iPhone near the reader Apple Pay automatically activate. 

Is there's such a way to configure the board to read raid card only and not scanning for NFC?

lesterlo, 2020-06-15:
This is the feature of Apple iPhone. When the iPhone detects any NFC/ RF wave, Apple Pay feature will activate automatically.

I think there is no way to disable this feature.

nutsmuggler, 2021-03-29:
The only way to prevent this is to use the PN532 in card emulation mode.
This way the iPhone will see it as a card, and read it correctly.

Cincinnatu, 2024-10-16:
> 亲爱的，只需要为我的应用程序读取 raid card。
> 
> 将 iPhone 连接到读卡器附近时，Apple Pay 会自动启动。
> 
> 有没有办法将板子配置为仅读取 RAID 卡而不扫描 NFC？
Hello Is the issue resolved now?

closed 2024-11-04 by Lesords