Seeed-Studio/PN532 #104: Not a issue but question?

akshar001, 2020-07-03:
For one of my application i will need both to emulate tag and use p2p ndef. at a time.
Can i do that? With current library or is is it even possible with using a chip?

Johannf78, 2020-07-03:
I am struggling with the same problem. Seems like technically it is possible. But have not been able to get it working. I think the phone / chips where more intended to read NFC cards, than to emulate a card. I am think now of using NFC to initiate the connection then go over to BLE.

DavidDHansson, 2024-03-06:
I am also in the same situation. I am looking into using threads. Thinking if a write has occurred within a sec or something, the loop can continue running, and then after it will continue checking write.

Cincinnatu, 2024-10-16:
Hello Is this issue currently resolved?

closed 2024-11-04 by Lesords