Seeed-Studio/PN532 #109: M5paper ESP32 Didn't find PN53x board

openmotion, 2021-01-24:
hello
i want to use the grove nfc v1.1 with my M5paper
the M5paper has a port C grove ( UART )

i can compile the code
but i have this error : 
`Didn't find PN53x board`

`#include <NfcAdapter.h>
#include <PN532/PN532/PN532.h>
#include <M5EPD.h>
#include "driver/gpio.h"
#include "driver/adc.h"

    #include <PN532/PN532_HSU/PN532_HSU.h>

 //HardwareSerial Serial1(1); 
    
    PN532_HSU pn532hsu(Serial1);
    NfcAdapter nfc(pn532hsu);
 
void setup(void) {
    M5.begin();
    Serial1.println("NDEF begin");
    delay(5000);
    Serial1.begin(9600);
    Serial1.println("NDEF Reader");
    nfc.begin();
}

void loop(void) {
    SERIAL.println("\nScan a NFC tag\n");
    if (nfc.tagPresent()) {
        NfcTag tag = nfc.read();
        tag.print();
    }
    //delay(5000);
     M5.update();
    delay(100);
}`

GeorgeFlorian, 2021-02-10:
I am having the some problem.

Did you solve it ?

openmotion, 2021-02-10:
> I am having the same problem.
> 
> Did you solve it ?

hello no :p

GeorgeFlorian, 2021-02-15:
> > I am having the same problem.
> > Did you solve it ?
> 
> hello no :p do you have the grove nfc v1.1

I am interested in integrating the RFID1356MIFARE with the ESP32-EVB an using it as a card reader. I don't think this library is compatible with the card reader firmware.

ABCDCreeper, 2021-07-10:
i have the same problem too.my pn532 is elechouse nfc moudle v3

Cincinnatu, 2024-10-16:
Has this issue been resolved yet? If not, I will take a look at it soon.

closed 2024-11-04 by Lesords