Seeed-Studio/PN532 #111: PN532 Card Emulation Problem

freeridre, 2021-04-26:
Hi There!

I would like to use my PN532 as a NFC tag that has some additional data which my Android Phone app could read and save. Unfortunatelly android does not recognize emulated tag, as you can see it.
I use emulate_tag_ndef from the examples.
Here is the exact example that I used:
`#include "emulatetag.h"
#include "NdefMessage.h"
#include <SPI.h>
#include <PN532_SPI.h>
#include "PN532.h"
PN532_SPI pn532spi(SPI, 10);
EmulateTag nfc(pn532spi);
uint8_t ndefBuf[120];
NdefMessage message;
int messageSize;
uint8_t uid[3] = { 0x12, 0x34, 0x56 };
void setup()
{
  Serial.begin(115200);
  Serial.println("------- Emulate Tag --------");
  message = NdefMessage();
  //message.addUriRecord("http://www.elechouse.com");
  message.addTextRecord("Hi");
  messageSize = message.getEncodedSize();
  if (messageSize > sizeof(ndefBuf)) {
      Serial.println("ndefBuf is too small");
      while (1) { }
  }  
  Serial.print("Ndef encoded message size: ");
  Serial.println(messageSize);
  message.encode(ndefBuf);  
  // comment out this command for no ndef message
  nfc.setNdefFile(ndefBuf, messageSize);  
  // uid must be 3 bytes!
  nfc.setUid(uid);  
  nfc.init();
}

void loop(){
    // uncomment for overriding ndef in case a write to this tag occured
    //nfc.setNdefFile(ndefBuf, messageSize);    
    // start emulation (blocks)
    nfc.emulate();        
    // or start emulation with timeout
    if(!nfc.emulate(1000)){ // timeout 1 second
      Serial.println("timed out");
    }    
    // deny writing to the tag
    // nfc.setTagWriteable(false);    
    if(nfc.writeOccured()){
       Serial.println("\nWrite occured !");
       uint8_t* tag_buf;
       uint16_t length;       
       nfc.getContent(&tag_buf, &length);
       NdefMessage msg = NdefMessage(tag_buf, length);
       msg.print();
    }
    delay(1000);
}`

And here is my array which is inside of the emulatetag.cpp
`uint8_t` command[] = {
      PN532_COMMAND_TGINITASTARGET,
      0x0,                  // MODE: 0x04 = PICC only, 0x01 = Passive only (0x02 = DEP only)

      // MIFARE PARAMS
      0x08, 0x00,         // SENS_RES (seeeds studio set it to 0x04, nxp to 0x08)
      0x00, 0x00, 0x00,   // NFCID1t    (is set over sketch with setUID())
      0x60,               // SEL_RES    (0x20=Mifare DelFire, 0x60=custom)

[... 257 lines cut ...]

        0, // length of general bytes
        0  // length of historical bytes
  };`

**Then here is my Serial Output:**
`22:50:02.335 -> ------- Emulate Tag --------
22:50:02.335 -> Ndef encoded message size: 9
22:50:02.335 -> SAMConfig
22:50:02.335 -> write:  14 01 14 01
22:50:02.335 -> read:   15
22:50:02.335 -> write:  8C 05 04 00 12 34 56 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
`
As you can see, nothing happends...

**Here is what I captured with Android S20 with NXP Tag info app:**
[image]
**Here is what I captured with Android S20 with NFC Tools pro app:**
It's same as in this question #88 
[image]

I also tried the whole thing with an Android S8 and it behaves similar to Andoird S20.
I hope somebody can help me.

Thank you so much for your helping in advance!

cino1001, 2021-05-17:
Hi,
I understood that we have to initiate the PN532  as soon as we received the status code 0x29 from TgGetData.
https://stackoverflow.com/questions/24777810/the-pn532-configured-as-target-has-been-released-by-its-initiator
so we have to modify the pn532.cpp - tgGetData() and , emulatetag.cpp - emulate()
but I'm stuck now don't really know how  to do that

mrdc, 2021-06-15:
@freeridre 
Hi, I'm having the same issue with PN532 (connected via SPI): 
when using this:
```
  PN532_COMMAND_TGINITASTARGET,
      0x0,                  // MODE: 0x04 = PICC only, 0x01 = Passive only (0x02 = DEP only)

      // MIFARE PARAMS
      0x08, 0x00,         // SENS_RES (seeeds studio set it to 0x04, nxp to 0x08)
      0x00, 0x00, 0x00,   // NFCID1t    (is set over sketch with setUID())
      0x60,               // SEL_RES    (0x20=Mifare DelFire, 0x60=custom)

      // FELICA PARAMS
      0x01, 0xFE,         // NFCID2t (8 bytes)
      0xA2, 0xA3, 0xA4,
      0xA5, 0xA6, 0xA7,
      0xC0, 0xC1, 0xC2,
      0xC3, 0xC4, 0xC5,
      0xC6, 0xC7, 0xFF,
      0xFF,
      0xAA, 0x99, 0x88, //NFCID3t (10 bytes)
      0x77, 0x66, 0x55, 0x44,
      0x33, 0x22, 0x11,

      0x00, // length of general bytes
      0x00  // length of historical bytes
```
I get this in output:
```
Ndef encoded message size: 29
SAMConfig
write:  14 01 14 01
read:   15
write:  8C 00 08 00 12 34 56 60 01 FE A2 A3 A4 A5 A6 A7 C0 C1 C2 C3 C4 C5 C6 C7 FF FF AA 99 88 77 66 55 44 33 22 11 00 00
tgInitAsTarget failed or timed out!write:  8C 00 08 00 12 34 56 60 01 FE A2 A3 A4 A5 A6 A7 C0 C1 C2 C3 C4 C5 C6 C7 FF FF AA 99 88 77 66 55 44 33 22 11 00 00
tgInitAsTarget failed or timed out!write:  8C 00 08 00 12 34 56 60 01 FE A2 A3 A4 A5 A6 A7 C0 C1 C2 C3 C4 C5 C6 C7 FF FF AA 99 88 77 66 55 44 33 22 11 00 00
tgInitAsTarget failed or timed out!write:  8C 00 08 00 12 34 56 60 01 FE A2 A3 A4 A5 A6 A7 C0 C1 C2 C3 C4 C5 C6 C7 FF FF AA 99 88 77 66 55 44 33 22 11 00 00
tgInitAsTarget failed or timed out!write:  8C 00 08 00 12 34 56 60 01 FE A2 A3 A4 A5 A6 A7 C0 C1 C2 C3 C4 C5 C6 C7 FF FF AA 99 88 77 66 55 44 33 22 11 00 00
```
and when using 
```
        PN532_COMMAND_TGINITASTARGET,
        5,                  // MODE: PICC only, Passive only

        0x04, 0x00,         // SENS_RES
        0x00, 0x00, 0x00,   // NFCID1
        0x20,               // SEL_RES

        0,0,0,0,0,0,0,0,
        0,0,0,0,0,0,0,0,   // FeliCaParams
        0,0,

        0,0,0,0,0,0,0,0,0,0, // NFCID3t

        0, // length of general bytes
        0  // length of historical bytes
```
I see this in output:
```
SAMConfig
write:  14 01 14 01
read:   15
write:  8C 05 04 00 12 34 56 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
tgInitAsTarget failed or timed out!write:  8C 05 04 00 12 34 56 20 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```

When I use PN532 for read/write it works fine.

vualeks, 2021-08-06:
Hmm, in my case it's working with both values for mode (0x00 and 0x05) with Android, but I would say with 0x05 slightly better, gets recognized more often.

On the other hand iPhone 8 doesn't work at all. Tried all modes, and many variations, but it seems that my iPhone just doesn't get through to this module.

You can find other IDs liste here http://nfc-tools.org/index.php/ISO14443A
I used Mifare DESFire and other combinations and most worked fine with Android, but iPhone no go.

mywalkb, 2022-06-12:
Emulation on PN532 is configured with both technology NfcA and NfcF. When I try to read the emulated tag with another PN532 and libnfc, it detect 2 tag 1 ISO14443 and 1 Felica (all zeros).
In my fork mywalkb/PN532 I emulated a tag as libnfc/examples/nfc-emulate-forum-tag2.c and works better on android is still not perfect.

masssya, 2023-08-19:
Hi, I also need to emulate an nfc tag using the PN532 module and send an ndef signal, when loading your code, I get a lot of errors that I can't fix in any way. We even wrote additional code for the Emulate Tag function using Cheat gpt, however, when correcting some errors, others appeared and so on. Do you think the problem may be in the wrong libraries or something else?

These are my mistakes: 
C:\Users\Max\Documents\Arduino\all_model_easy\all_model_easy.ino:6:1: error: 'EmulateTag' does not name a type
C:\Users\Max\Documents\Arduino\all_model_easy\all_model_easy.ino: In function 'void setup()':
C:\Users\Max\Documents\Arduino\all_model_easy\all_model_easy.ino:27:3: error: 'nfc' was not declared in this scope
C:\Users\Max\Documents\Arduino\all_model_easy\all_model_easy.ino: In function 'void loop()':
C:\Users\Max\Documents\Arduino\all_model_easy\all_model_easy.ino:37:5: error: 'nfc' was not declared in this scope

exit status 1

Compilation error: 'EmulateTag' does not name a type

Dermiste, 2024-02-19:
Has anyone looked into this and managed to make this work properly ? (will post details of my scenario if required)

whywilson, 2024-09-27:
> Has anyone looked into this and managed to make this work properly ? (will post details of my scenario if required)

I made it works with Python and PN532 on HSU mode. [PN532 - Python](https://github.com/whywilson/pn532-python)

Cincinnatu, 2024-10-16:
Hello, I will try to solve this problem soon. Could you please provide a detailed description of the issue? This will help me to reproduce and resolve the issue.

Lesords, 2024-11-04:
Hello,

I\'m going to close this issue, feel free to re-open it if you have any other questions.

[Autocommentary]

closed 2024-11-04 by Lesords

lwin12, 2024-12-07:
Hi, I somehow managed to get it to emulate it to act as an NFC Tag and was able to successfully read it on my IPhone. I will post the working code tomorrow.

AnonO6, 2025-03-26:
> Hi, I somehow managed to get it to emulate it to act as an NFC Tag and was able to successfully read it on my IPhone. I will post the working code tomorrow.

Hi @lwin12 can you provide you code solution. Would help a lot, thanks!

lwin12, 2025-04-10:
```
#include <SPI.h>
#include <PN532_SPI.h>
#include <PN532.h>
#include "emulatetag.h"

// Define SPI Pins
#define PN532_CS 5 // Chip Select pin for PN532

PN532_SPI pn532spi(SPI, PN532_CS);
PN532 nfc(pn532spi);
EmulateTag emulateTag(pn532spi);

// Buffers and Defaults
uint8_t ndefBuf[128]; // Limited to 128 bytes due to PN532 hardware constraint
uint8_t uid[7] = { 0x04, 0x8A, 0x3F, 0x01, 0x02, 0x03, 0x04 }; // 7-byte UID

// Shared variables
char currentMessage[120] = "Hello World!";
bool messageUpdated = false; // Flag to indicate a new message
SemaphoreHandle_t messageMutex; // Mutex to protect shared variables

void setup() {
    Serial.begin(115200);
    Serial.println("---- NFC Tag Emulation with Immediate Updates ----");

    // Initialize PN532
    Serial.println("Initializing PN532...");
    nfc.begin();
    uint32_t versiondata = nfc.getFirmwareVersion();

    if (!versiondata) {
        Serial.println("Error: Could not detect PN532. Check wiring and SPI configuration.");
        while (1); // Halt execution if PN532 is not detected
    }

    Serial.println("PN532 detected!");
    nfc.SAMConfig();
    Serial.println("PN532 is ready for NFC operation.");

    // Create a mutex for thread-safe access
    messageMutex = xSemaphoreCreateMutex();

    // Start tasks on different cores
    xTaskCreatePinnedToCore(serialInputTask, "SerialInputTask", 4096, NULL, 1, NULL, 0); // Run on Core 0
    xTaskCreatePinnedToCore(emulationTask, "EmulationTask", 4096, NULL, 1, NULL, 1);    // Run on Core 1
}

void loop() {
    // Empty, as tasks handle everything
}

// Task to handle Serial input (Core 0)
void serialInputTask(void* parameter) {
    while (true) {
        if (Serial.available() > 0) {
            String input = Serial.readStringUntil('\n'); // Read input until newline
            input.trim(); // Remove leading/trailing whitespace

            if (input.length() > 0) {
                Serial.println("Received new message: " + input);

                // Safely update the shared message
                xSemaphoreTake(messageMutex, portMAX_DELAY); // Lock mutex
                input.toCharArray(currentMessage, sizeof(currentMessage)); // Copy input
                messageUpdated = true; // Set update flag
                xSemaphoreGive(messageMutex); // Unlock mutex
            }
        }

        vTaskDelay(10 / portTICK_PERIOD_MS); // Allow other tasks to run
    }
}

// Task to handle NFC emulation with immediate updates (Core 1)
void emulationTask(void* parameter) {
    while (true) {
        char messageToEmulate[120];
        bool updated = false;

        // Safely copy the message and check if it's updated
        xSemaphoreTake(messageMutex, portMAX_DELAY); // Lock mutex
        strncpy(messageToEmulate, currentMessage, sizeof(messageToEmulate));
        updated = messageUpdated;
        messageUpdated = false; // Reset update flag
        xSemaphoreGive(messageMutex); // Unlock mutex

        // If there's an update, refresh the NFC emulation
        if (updated == true) {
            Serial.println("Updating NFC tag with new message...");
            updateNdefMessage(messageToEmulate);
        }

        // Continuously emulate NFC tag (no timeout)
        bool success = emulateTag.emulate(500); // Emulate for 500ms and check updates again
        if (success == true) {
            Serial.println("Tag was successfully read!");
        }

        vTaskDelay(10 / portTICK_PERIOD_MS); // Allow other tasks to run
    }
}

// Function to update the NDEF message
void updateNdefMessage(const char* text) {
    const int maxPayloadLength = 119; // Maximum payload length for 126-byte storage
    char truncatedText[maxPayloadLength + 1]; // +1 for null terminator

    // Truncate the text if it exceeds the limit
    strncpy(truncatedText, text, maxPayloadLength);
    truncatedText[maxPayloadLength] = '\0'; // Ensure null termination

    int textLength = strlen(truncatedText);
    const int ndefHeaderSize = 7; // Fixed NDEF header size

    // Construct NDEF Text Record Header
    ndefBuf[0] = 0xD1; // NDEF header
    ndefBuf[1] = 0x01; // Type Length
    ndefBuf[2] = textLength + 3; // Payload Length
    ndefBuf[3] = 0x54; // Type
    ndefBuf[4] = 0x02; // Language code length
    ndefBuf[5] = 'e';  // Language code byte 1
    ndefBuf[6] = 'n';  // Language code byte 2

    memcpy(&ndefBuf[7], truncatedText, textLength);

    emulateTag.setUid(uid);
    emulateTag.setNdefFile(ndefBuf, textLength + ndefHeaderSize);

    Serial.println("NDEF message updated!");
}

```

lwin12, 2025-04-10:
It is still a bit buggy, but it should work. @AnonO6

AnonO6, 2025-04-11:
> It is still a bit buggy, but it should work.
> Sorry for the late reply as I have been busy with school projects.
> @AnonO6 

Thanks a lot @lwin12

baorepo, 2025-07-29:
@lwin12  @sadsultan2  Could you create a new issue to continue the discussion on this?

lwin12, 2025-07-29:
Feel free to drop me a PM at my discord handle if you require the correct library and source code. I am currently very busy with other projects.

> [@lwin12](https://github.com/lwin12) [@sadsultan2](https://github.com/sadsultan2) Could you create a new issue to continue the discussion on this?