Seeed-Studio/PN532 #113: SPI Wiring Not working on d1 mini

6000850, 2021-05-11:
Hello
This wiring is SPI, but after uploading the ReadTag.ino code, the message "Didn't find PN53x board" is displayed.

My Board : 
`Wemos D1 Mini`

My Wiring :
```
SCK => D5
MISO => D6
MOSI => D7
SS => D3
VCC => 3.3V
GND => GND
```

ReadTag.ino

```

#include <SPI.h>
#include <PN532_SPI.h>
#include <PN532.h>
#include <NfcAdapter.h>

PN532_SPI pn532spi(SPI, 0);
NfcAdapter nfc = NfcAdapter(pn532spi);

void setup(void) {
    Serial.begin(9600);
    Serial.println("NDEF Reader");
    nfc.begin();
}

void loop(void) {
    Serial.println("\nScan a NFC tag\n");
    if (nfc.tagPresent())
    {
        NfcTag tag = nfc.read();
        tag.print();
    }
    delay(5000);
}
```

6000850, 2021-05-12:
Hello
No answer to this question?

6000850, 2021-05-19:
SoftwareSerial worked! but SPI NOT Working!! why!?

Cincinnatu, 2024-10-16:
Hello, has this issue been resolved? If not, please provide more information.

closed 2024-11-04 by Lesords