Seeed-Studio/PN532 #119: Tag emulation not working for iPhone

vualeks, 2021-08-06:
Hi,

I have this PN532 module connected with arduino and wanted to try tag emulation. With these params for initialization emulatetag.cpp:

`uint8_t command[] = {`
 `     PN532_COMMAND_TGINITASTARGET,`
`      0x04,                  // MODE: 0x04 = PICC only, 0x01 = Passive only (0x02 = DEP only)`

   `   // MIFARE PARAMS`
 `     0x44, 0x03,         // SENS_RES (seeeds studio set it to 0x04, nxp to 0x08)`
  `    0xdc, 0x44, 0x20,   // NFCID1t    (is set over sketch with setUID())`
   `   0x20,               // SEL_RES    (0x20=Mifare DelFire, 0x60=custom)`
`
      // FELICA PARAMS
      0x01, 0xfe, // NFCID2T MUST START WITH 01fe - FELICA PARAMS - POL_RES
      0xa2, 0xa3, 0xa4, 0xa5, 0xa6, 0xa7, 0xc0,
      0xc1, 0xc2, 0xc3, 0xc4, 0xc5, 0xc6, 0xc7, // PAD
      0xff, 0xff,                               // SYSTEM CODE
      0xaa, 0x99, 0x88, 0x77, 0x66, 0x55, 0x44,
      0x33, 0x22, 0x11, 0x01, 0x00, // NFCID3t MAX 47 BYTES ATR_RES
      0x0d, 0x52, 0x46, 0x49, 0x44, 0x49, 0x4f,
      0x74, 0x20, 0x50, 0x4e, 0x35, 0x33, 0x32// length of historical bytes
  };`
  
  Android tag detection and writing works ok. I've tried other combinations for mode (0x00, 0x02, 0x04) and all of them work ok for Android, but my iPhone 8 just doesn't recognize it. 
  I've tried many tag types from here http://nfc-tools.org/index.php/ISO14443A Most worked ok for Android, none for iPhone. Tried with NFC tools app and others, even my own code for writing tags that I could debug, but it simply doesn't see anything. Also, debugging PN532 library didn't show any incoming signals/commands like it did with Android. With iPhone just nothing. It seems to me that it could be some hardware problem with their communication (module and iPhone), maybe weak antenna signal?
  
 Anyone has some suggestions what to try?

freeridre, 2021-08-21:
Hi! Try this in emulatetag.cpp
```
  uint8_t command[] = {
      PN532_COMMAND_TGINITASTARGET,
      0x01,                  // MODE: 0x04 = PICC only, 0x01 = Passive only (0x02 = DEP only)

      // MIFARE PARAMS
      0x04, 0x00,         // SENS_RES (seeeds studio set it to 0x04, nxp to 0x08)
      0x00, 0x00, 0x00,   // NFCID1t    (is set over sketch with setUID())
      0x20,               // SEL_RES    (0x20=Mifare DelFire, 0x60=custom)

      // FELICA PARAMS
      0x01, 0xFE,         // NFCID2t (8 bytes) https://github.com/adafruit/Adafruit-PN532/blob/master/Adafruit_PN532.cpp FeliCa NEEDS TO BEGIN WITH 0x01 0xFE!
      0xA2, 0xA3, 0xA4,
      0xA5, 0xA6, 0xA7,
      0xC0, 0xC1,         // PAD (8 bytes)
      0xC2, 0xC3, 0xC4, 
      0xC5, 0xC6, 0xC7,   
      0xFF, 0xFF,         // System code (2 bytes)

      0xAA, 0x99, 0x88,   // NFCID3t (10 bytes)
      0x77, 0x66, 0x55, 0x44,
      0x33, 0x22, 0x11,

      0x00, // length of general bytes
      0x00  // length of historical bytes
  };
```

freeridre, 2021-08-21:
I have and Iphone 12 pro max, and I can read and write Ndef to Ardunio.

vualeks, 2021-08-22:
No use, I've tried that too.

It seams that it is some deeper problem. My iPhone 8 doesn't react or want to communicate at all with the module but I've tried my friends iPhone 12 and it did communicate, but gave some tag reading error. I didn't have my laptop at hand to debug what was the problem, but for sure it was software related and it could probably be fixed to work with iPhone 12.

The thing is that it doesn't behave the same way at all with iPhone 8, and if that's the case then there is some bigger underlying problem.

closed 2024-11-04 by Lesords