Seeed-Studio/PN532 #120: Tag Emulation Write Occurred!

tmckenna-1, 2021-08-20:
Hello, I am trying to use an Adafruit PN532 in Emulation mode to trigger a locking mechanism. Basically, the idea is when a write to a tag occurs via a mobile device, the PN532 signals an Arduino which triggers the lock to open. I am running into an issue however, where once a write occurs to the module, that one write triggers the Arduino for every loop (i. e. the lock will open even when a write does not occur). Any idea on how I can fix this? My code is attached below:

//libraries
#include "emulatetag.h"
#include "NdefMessage.h"

#include <SPI.h>
#include <PN532_SPI.h>
#include "PN532.h"

//defines RELAY pin - can be whatever pin necessary
int RELAY = 5;

//initializes PN 532 with SS pin denoted as pin 10
PN532_SPI pn532spi(SPI, 10);
EmulateTag nfc(pn532spi);

//Sets limit for message at 120 bytes
//Initializes message size
uint8_t ndefBuf[120];
NdefMessage message;
int messageSize;

//Creates 3 byte Unique Identifier for emulated tag
uint8_t uid[3] = {0x12, 0x34, 0x56};

void setup()
{
  Serial.begin(115200);
  Serial.println("------- Emulate Tag --------");

  pinMode(RELAY, OUTPUT);
 
  //Creates NDEF message
  message = NdefMessage();

 //Record of message - can add anything here (text, URL, etc.)
  message.addUriRecord("");
  messageSize = message.getEncodedSize();

 //Ensures message is smaller than 120 bytes
  if (messageSize > sizeof(ndefBuf))
  {
    Serial.println("ndefBuf is too small");
    while (1)
    {
    }
  }

  Serial.print("Ndef encoded message size: ");
  Serial.println(messageSize);

  //encodes message to board
  message.encode(ndefBuf);

  // comment out this command for no ndef message
  nfc.setNdefFile(ndefBuf, messageSize);

  // uid must be 3 bytes
  nfc.setUid(uid);

  nfc.init();
}

void loop()
{
  digitalWrite(RELAY, LOW);

  //overrides ndef in case a write to this tag occured
  //nfc.setNdefFile(ndefBuf, messageSize);

  Serial.println("Emulate");

  // start emulation with timeout
  if(!nfc.emulate(3000)){ // timeout 3 second
      Serial.println("timed out");
    }

  //when the tag is written by some outside entity (in this case a mobile phone)
  if (nfc.writeOccured())
  {
    Serial.println("\nWrite occured !");
    uint8_t *tag_buf;
    uint16_t length;

    //signals relay for 0.5 seconds, releasing lock
    digitalWrite(RELAY, HIGH);
    delay(500);
  }
}

tmckenna-1, 2021-08-20:
Update: As a temporary fix, I've added in some code so that the Arduino resets itself when a write has occurred. This works in clearing the write to the NFC, but I'd still like a cleaner solution if anyone has any suggestions.

Cincinnatu, 2024-10-16:
Can you provide some more information and your new code? I will take a look at the issue soon.