Seeed-Studio/PN532 #125: PN532 Emulate not working randomly

mdjavedakhtar, 2022-03-31:
Hi
i have tried emulating a card with PN532 module. it works like 1 out of 20 times

i debug i get tgInitAsTarget Failed or timeout

if i keep the phone on the pn532 with 1 sec timeout in sketch after about 20 retry i get data on phone. then again it stops.

i am using SPI connection.

mywalkb, 2022-06-12:
Try to see if my changes mywalkb/PN532 improve your experience.
nfc.emulate use timeout 0, it don't give time to read response from chip, is not infinite timeout.

jlkalberer, 2023-09-04:
@mywalkb - Can you explain what changes you're making in your branch? 
There is a lot going on there and I'd like to understand the changes before I try to use your solution as I'm sure I'd only need parts of it.

Cheers!

mywalkb, 2023-09-04:
> @mywalkb - Can you explain what changes you're making in your branch? 

it's been over a year it's hard to remember all the changes and why I made them.
With my changes I tried to implement libnfc's functions which works very well with PN532.
I report the comment in my first commit:

```
- change include path
- fix some functions with readresponse when it's return 0 length (is not an error). Some other todo
- add setTimeoutComm for change timeout for command inCommunicateThru
- improvement function inCommunicateThru manage buffer len 0 so we can ignore response from TAG
- improvement function startPassiveTargetIDDetection manage better paramenters for all cardbaudrate
```

>  The other commits on your branch looked like it was for supporting callbacks which I don't want or need.

you can fork and remove that commit if you don't want it. When you emulate a tag without callback, how can you know when you receive data? Polling is not a good method in my opinion.

jlkalberer, 2023-09-04:
Thanks @mywalkb 

[I took a lot of what you did and put it in this branch](https://github.com/Brewskey/Brewskey.Device/pull/6/files)
I'll test this out and let you know if it works for me. If it does, I'll make a smaller patch and put a PR up on this repo.

---

> Polling is not a good method in my opinion.

Yeah, I'd generally agree with this but my code runs in a loop/state machine so a callback doesn't help me too much :)

Cincinnatu, 2024-10-16:
Hello！Has this issue been resolved yet? If not, I will take a look at it soon.

mdjavedakhtar, 2024-10-20:
Still the issue is there

ackPeng, 2025-01-07:
@mdjavedakhtar Hello, I tested it using FeliCa_card_detection. After turning on the phone, I put the card on my phone and it can be recognized.

jlkalberer, 2025-01-07:
I haven't been able to verify this yet but it I used another iOS app that could scan the emulated tag. I reached out to the author for details.

Basically, they had two reader implementations:
* [NFCTagReaderSession](https://developer.apple.com/documentation/corenfc/nfctagreadersession) - this is a newer, higher level API. It is not working for the scans.
* [NFCNDEFReaderSession](https://developer.apple.com/documentation/corenfc/nfcndefreadersession) - this works for scanning the emulated tag.

I'm not sure if he manually wrote the protocol for reading the tag with `NFCNDEFReaderSession` or not but I know for a fact that it is working.

ackPeng, 2025-01-08:
@jlkalberer My phone is an Android system, and every scan is successful. I haven’t tried it on IOS yet. I’ll give it a try on IOS.

ackPeng, 2025-01-08:
@jlkalberer I also tested it on an Apple mobile phone and it can also be detected.

ackPeng, 2025-01-15:
This issue will be closed soon. If you have further questions, please create another issue.

closed 2025-01-15 by ackPeng