Seeed-Studio/PN532 #137: Can't enable debug on esp32.

TECH7Fox, 2023-01-08:
Using a ESP32 Dev Module.
If I uncomment `#define DEBUG` to enable debug I get this:

```
In file included from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/stdio.h:47:0,
                 from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:27,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\PN532_debug.h:6,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\emulatetag.cpp:10:
C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/sys/reent.h:463:10: error: #if with no expression
In file included from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/stdio.h:47:0,
                 from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:27,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\PN532_debug.h:6,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\llcp.cpp:3:
C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/sys/reent.h:463:10: error: #if with no expression
In file included from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/stdio.h:47:0,
                 from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:27,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\PN532_debug.h:6,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\mac_link.cpp:3:
C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/sys/reent.h:463:10: error: #if with no expression
In file included from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/stdio.h:47:0,
                 from C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:27,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\PN532_debug.h:6,
                 from C:\Users\Jordy\Documents\Arduino\libraries\PN532\snep.cpp:3:
C:\Users\Jordy\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/newlib/sys/reent.h:463:10: error: #if with no expression

exit status 1

Compilation error: exit status 1
```

Anyone else got this?

qbonenfant, 2023-02-20:
Hi,
I got the same issue on a CubeCell HTCC-AB02.
In my case, there was a conflict with the board library, which also use a DEBUG statement in its debug.h.
Using such common names in a library is poor practice and should be fixed ASAP (on both side really).

To solve the issue, you can modify the redundancy checks in PN532_debug.h

Here is what my file looks like:

`#ifndef __DEBUG_PN532_H_
#define __DEBUG_PN532_H_

#define DEBUG_532

#include "Arduino.h"

#ifdef DEBUG_532
#define DMSG(args...)       Serial.print(args)
#define DMSG_STR(str)       Serial.println(str)
#define DMSG_HEX(num)       Serial.print(' '); Serial.print((num>>4)&0x0F, HEX); Serial.print(num&0x0F, HEX)
#define DMSG_INT(num)       Serial.print(' '); Serial.print(num)
#else
#define DMSG(args...)
#define DMSG_STR(str)
#define DMSG_HEX(num)
#define DMSG_INT(num)
#endif

#endif
`

This fix may need to be propagated to other files if there is an update of the library, but should work for now.
I should really make a PR to fix this, but don't have time RN (maybe later... if i remember)

alireza-tabatabaee, 2023-05-26:
As a quick and dirty fix you can comment out 
#define SERIAL  0x0
in line 50 of Arduino.h which is located at %localappdata%/Arduino15/packages/esp32/hardware/esp32/2.0.8/cores/esp32/Arduino.h

Cincinnatu, 2024-10-16:
Hello, I will check soon whether this issue still exists. If it does, I will work on fixing it.

Lesords, 2025-01-06:
Hello，

I'm very sorry to have kept you waiting so long.

Do you still have this problem now?

If so, can you try the latest code?

If the latest code also has your problem, can you tell me the board model you are using and the corresponding sketch?

qbonenfant, 2025-01-06:
Hi,
I do have a project that will require a PN532 pretty soon. I will keep you updated.

Lesords, 2025-01-07:
> Hi, I got the same issue on a CubeCell HTCC-AB02. In my case, there was a conflict with the board library, which also use a DEBUG statement in its debug.h, with a IFNDEF DEBUG to avoid redundancy. Using such common names in a library is poor practice and should be fixed ASAP (on both side really).
> 
> To solve the issue, you can modify the redundancy checks in PN532_debug.h
> 
> Here is what my file looks like:
> 
> ```
> #ifndef __DEBUG_PN532_H_
> #define __DEBUG_PN532_H_
> 
> #define DEBUG_532
> 
> #include "Arduino.h"
> 
> #ifdef DEBUG_532
> #define DMSG(args...)       Serial.print(args)
> #define DMSG_STR(str)       Serial.println(str)
> #define DMSG_HEX(num)       Serial.print(' '); Serial.print((num>>4)&0x0F, HEX); Serial.print(num&0x0F, HEX)
> #define DMSG_INT(num)       Serial.print(' '); Serial.print(num)
> #else
> #define DMSG(args...)
> #define DMSG_STR(str)
> #define DMSG_HEX(num)
> #define DMSG_INT(num)
> #endif
> 
> #endif
> ```
> 
> I hope it will work for you too. This fix may need to be propagated to other files if there is an update of the library, but should work for now. I should really make a PR to fix this, but don't have time RN (maybe later... if i remember)

I don't have this board

Is the problem you encounter only the DEBUG macro conflict?

Does this code snippet solve all your problems?

Lesords, 2025-01-07:
By the way, we welcome your PRs

ackPeng, 2025-01-15:
This issue will be closed soon. If you have further questions, please create another issue.

closed 2025-01-15 by ackPeng