Seeed-Studio/PN532 #139: PN532 module not detecting as emulated tag with Nano 33 BLE to Android

liswifi, 2023-03-05:
It hasn't been detecting the PN532 at all while using the emulate_tag_ndef.ino example from this library. 

I find it strange that it compiles but doesn't detect the module. Lots of 'unused variable' and 'redefined variable' warnings appear during compilation but I'm unsure why. Here is a pastebin of the compile: [SeedStudio PN532 emulate_tag_ndef.ino](https://pastebin.com/KGHNu9ha)

Jenibluere, 2023-03-18:
Hey, if you follow the steps outlined in this issue here, you can get it to work: https://github.com/Seeed-Studio/PN532/issues/88

jlkalberer, 2023-09-04:
@Jenibluere - what steps are you suggesting? There are many different ideas on how to solve in that thread.

Cincinnatu, 2024-10-16:
Has this issue been resolved? If not, I will try to solve the problem soon.

jlkalberer, 2024-10-16:
Yes, tag emulation is basically broken for iOS and Android at this point.

Every once in a while I can get a successful read (Android) but in general it's broken.

Cincinnatu, 2024-10-17:
It seems that some systems and phones do not support it.

Lesords, 2024-11-04:
Hello,

I\'m going to close this issue, feel free to re-open it if you have any other questions.

[Autocommentary]

closed 2024-11-04 by Lesords

jlkalberer, 2024-11-04:
No, it's not that phones don't support this chip. The issue is that this code doesn't do the full handshake with error handling so most of the time the read fails. I've looked at logcat and I've seen a ton of errors since the code is missing parts of the protocol.