Seeed-Studio/PN532 #141: Support for ISO 14443-4 tag

tsamppa, 2023-11-01:
I desperately need ISO 14443-4 card type to work (dual mode Fundan FM1280 card), but I am not an expert enough to be able to implement the support to this library.

I am using ESP32 Dev Module (KMP Electronics PRODINo ESP32 v1) with the Seeed-Studio Grove-NFC PN532 module with hardware Serial connected with Grove, so the PN532_HSU interface.

At the moment Mifare classic cards work ok, but when presented with the ISO 14443-4 compatible card nothing happens. The receive() function in PN532_HSU retuns constantly 0 after the timeout.

What kind of work would it be to add support for ISO 14443-4 cards to this library?

Or are there other libraries that already support ISO 14443-4, which I could try to use with Grove and hardware serial in HSU mode?

Or what other options I have?

Cincinnatu, 2024-10-16:
Hello！Has this issue been resolved? If not, I will take a look at the issue soon.

tsamppa, 2024-10-25:
I never got the PN532 to work with the card, so I switched to PN5180 that works like a charm.

Cincinnatu, 2024-10-28:
> I never got the PN532 to work with the card, so I switched to PN5180 that works like a charm.

I will look into this issue. Could you provide more information?

tsamppa, 2024-10-29:
Well, most other cards were recognized by the PN532 normally, except the Fundan FM1280 card, that is used as employee tax identification card on constuction sites in Finland (called Valttikortti in finnish). It is a dual mode card that supports ISO 14443-4 standard. The PN532 never recognized the card in vincinity.

The PN5180 recognized that card flawlessly.

Cincinnatu, 2024-12-12:
Hello,Perhaps you can provide your code for PN532, since I see that this library includes support for functions of ISO14443A.

ackPeng, 2024-12-26:
This issue will be closed. If you still have questions, please open a new issue.

closed 2024-12-26 by ackPeng