Seeed-Studio/PN532 #144: bibliothèque invalide

stephcache, 2024-05-06:
Bonjour, j'ai télécharger votre bibliothèque pour ajouter au logiciel Arduino et il indique qu'elle n'est pas valide

Pr77Pr77, 2024-06-30:
Use this repository instead: https://github.com/Seeed-Studio/Seeed_Arduino_NFC
But use the examples from this repository, because in Seeed_Arduino_NFC they aren't working. But you have to add
`#include <NfcAdapter.h>`
and put included files in these signs <> and add `PN532/PN532`
E. g.: 
Replace...
```
#include <SPI.h>
#include <PN532_SPI.h>
#include "PN532.h"
```
with...
```
#include <NfcAdapter.h>

#include <SPI.h>
#include <PN532/PN532_SPI/PN532_SPI.h>
#include <PN532/PN532/PN532.h>
```

Cincinnatu, 2024-10-16:
> Bonjour， j'ai télécharger votre bibliothèque pour ajouter au logiciel Arduino et il indique qu'elle n'est pas valide

This seems to be able to solve the problem: https://github.com/Seeed-Studio/PN532/issues/144#issuecomment-2198488665.

closed 2024-11-04 by Lesords