Seeed-Studio/PN532 #149: Cannot create PN532_SPI instance.

aloyen07, 2024-11-20:
Hello!
I'm have this code:
```
#define NFC_INTERFACE_SPI

#include <SPI.h>
#include <PN532_SPI.h>
#include "PN532.h"

SPIClass spiClass();
PN532_SPI pn532spi(SPI, 6);
PN532 nfc(pn532spi);

void begin() {
  nfc.begin();
}

void loop() {

}
```
But it not works. Compilator throws this error:
```
/tmp/cc85nja4.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_eee.ino.cpp.o.2084':
<artificial>:(.text.startup+0x62): undefined reference to `PN532_SPI::PN532_SPI(SPIClass&, unsigned char)'
/tmp/cc85nja4.ltrans0.ltrans.o: In function `main':
/home/aloyen/.arduino15/packages/arduino/hardware/avr/1.8.6/cores/arduino/main.cpp:43: undefined reference to `setup'
collect2: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1
```

How to fix it&

reganissu, 2024-11-23:
Hello, I have happened to solve this issue a while ago.  It was happening because newer arduino rules deprecated a syntax declaration used in SPI include file of that library, if memory serves me well.  Sorry for now I am not able to post my solution, only this hint to go toward solution.  I rember though that the solution was changing few lines of code regarding that deprecated syntax

sduponch, 2024-11-27:
Adding -DNFC_INTERFACE_I2C / -DNFC_INTERFACE_SPI to compiler.cpreprocessor.flags.esp32s3= in platform.txt fix the issue (Yes i have "#define NFC_INTERFACE_I2C" at start of my code).

BenEspinoza10, 2024-12-01:
@reganissu Im having the exactly same problem, have you remembered the solution?

sduponch, 2024-12-02:
Try replacing 
`#define NFC_INTERFACE_SPI `
by 
`constexpr int NFC_INTERFACE_I2C = 1;`

reganissu, 2024-12-02:
> @reganissu Im having the exactly same problem, have you remembered the solution?

Ok, I have found the workaround I have done and that made everything work for me.  I am not totally sure it is on topic for your issue, but I am posting the solution below and you can judge for yourself. 

In the PN532_SPI folder, in PN532_SPI.cpp file, in the void PN532_SPI::begin() function in the first 30 lines of code, you have to make the following mod, since SetDataMode and SetBitOrder have been deprecated a long ago (by arduino SPI library I think - not sure):
`    _spi->begin();
    _spi->beginTransaction(SPISettings(2000000, LSBFIRST, SPI_MODE0));
   // _spi->setDataMode(SPI_MODE0); // PN532 only supports mode0 // MOD SPISettings
    //_spi->setBitOrder(LSBFIRST); // 

//   SPISettings(2000000, LSBFIRST, SPI_MODE0);
`
as you can see I have commented out the deprecated declaration, using the "new" accepted SPIsettings declaration.
Hope this helps

sduponch, 2024-12-03:
The issue is that `#define NFC_INTERFACE_SPI` is not set when the compiler build the PN532 lib.

The #define in the ".ino" file seems out of scope : A workaround is to manually edit the compiler args by adding `-DNFC_INTERFACE_SPI` in the compiler command line (ex: by editing the "platform.txt" in the Arduino "hardware" directory).

Cincinnatu, 2024-12-11:
Hello，I just tested this issue and found that adding #include <PN532_SPI.cpp> allows it to run normally.

ackPeng, 2024-12-26:
This issue has been fixed and will be closed soon. If you still have any doubts, please reopen one.

closed 2024-12-26 by ackPeng