Seeed-Studio/PN532 #150: Android cannot recognize emulated card, while iPhone does

hhhtylerw, 2025-01-16:
Running the [emulate_tag_ndef](https://github.com/Seeed-Studio/PN532/tree/arduino/examples/emulate_tag_ndef) example, as is, I am able to emulate an nfc card with a single URL record. Using NFC Tools, I am able to both read and write the emulated card using my iPhone XR, however, I am unable to do the same with my Pixel 8. 

What could be causing this? 

For further context, my Pixel is capable of reading and writing cards (this is a physical card I wrote to previously):
[image]

This is what my iPhone reads off of the emulated card:
[image]

Perhaps the problem could have something to do with the lack of any standard "Technology" or other field in the emulated card? I'm just grasping at straws here. 

My setup:
PN532 over HSU
```
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = 
    https://github.com/Seeed-Studio/PN532.git
build_flags =
    -DNFC_INTERFACE_HSU
```

r8u, 2025-01-16:
Same issue here.

ackPeng, 2025-01-16:
@hhhtylerw  Hi, my phone is Redmi K70 Pro, it is an Android system, I used a tool that seems to be able to read its information,What NFC tool are you using?
[image]
[image]

ackPeng, 2025-01-16:
In addition, when I used it, the data reception was occasionally incomplete. After that, I put the card on the phone again and it was able to recognize all the data. Could it be caused by poor contact?

hhhtylerw, 2025-01-16:
@ackPeng This is the NFC app I am using. I believe it is the most popular one in the US. Normally, when you hold a phone near an NFC tag with a URL message, the phone will pop up with a link that you can click. I can not

ackPeng, 2025-01-17:
@hhhtylerw  Yes, when I put the NFC tag close to my phone without opening the app, a link will appear; I also downloaded your app and found that it can still read the card information. Maybe it’s because we use different chips? The picture shows the hardware I used and the results on the phone:

[image]
[image]
[image]
[image]

ackPeng, 2025-01-20:
@hhhtylerw okk,Welcome to your further feedback. This issue will close. If you have further questions, you can open a new one.

closed 2025-01-20 by ackPeng