Seeed-Studio/PN532 #155: Emulate tag with SPI on esp32s3

agekoda, 2025-04-30:
Hi there,
I'm trying to emulate an NDEF URI using a PN532 module connected to an ESP32-S3. The PN532 is wired to VSPI (SPI bus 3) with the following pinout:
MOSI: GPIO 35
MISO: GPIO 37
SCK: GPIO 36
SS/CS: GPIO 39
I'm using the example `emulate_tag_ndef.ino`, which includes the following code:
`#if 0
  #include <SPI.h>
  #include <PN532_SPI.h>
  #include "PN532.h"

  PN532_SPI pn532spi(SPI, 39);  // Example uses default SPI bus and my ss (gpio 39)
  EmulateTag nfc(pn532spi);`

However, this uses the default SPI bus, and I'm not sure how to configure it to use VSPI (bus 3) with my custom pinout. I’ve tried modifying the SPI object but haven’t had success.
How can I properly initialize the SPI bus with my pin configuration on the ESP32-S3?
Any help is greatly appreciated!

ackPeng, 2025-05-07:
@agekoda  You can refer to SPI.h to construct your own spi object, and then pass this object to PN532_SPI, so that you can use your own SPI

ackPeng, 2025-05-09:
This issue will be closed. If you have further questions, please open a new one.

closed 2025-05-09 by ackPeng